wills106/homeassistant-solax-modbus #507: [BUG] Solax X1 no cummincation since 2023.08.6 update

kevinwells22, 2023-08-24:
I was running version before 2023.08.6 with no issues, I upgraded to version 2023.08.6 yesterday, and i now have 'unavailable' on all sensors. I removed and re-installed this integration, but same issue.
I have this Inverter: Solax Power X 1-2 3.0Kw Single Phase Inverter.

wills106, 2023-08-24:
Do you get an error in the Log?

Which X1 are you connecting to? PV Only or a Gen3 or Gen4 Hybrid and how are you actually connecting to the Inverter?

If you are using a PocketWiFi have a look at the comments here in the [Readme](https://github.com/wills106/homeassistant-solax-modbus#support-modbus-over-rs485--tcp)

kevinwells22, 2023-08-25:
Thank you for your reply, I am indeed using the Pocket Wifi 3.0, I rebooted the Inverter, and communication all came back. Thank you.

kevinwells22, 2023-08-25:
Fixed.

closed 2023-08-25 by kevinwells22