wills106/homeassistant-solax-modbus #508: [BUG]? Meter Power Factor is missing a unit

dakaix, 2023-08-27:
**Describe the bug**
The sensor for Meter Power Factor does not have a unit of measurement set, this results in every change being recorded as a state (rather than displaying as a graph) and results in very significant churn in the Home Assistant logbook. 

**Mandatory details**
1. Integration Version: 2023.08.6
2. HA Version: 2023.8.4
3. Inverter Brand: Solis
4. Plugin Used: HACS
5. First 6 digits of Serial: 603015
6. Connection Method: TCP

[image]
[image]

wills106, 2023-08-29:
Just released [2023.08.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.7b2) does it look correct now?

wills106, 2023-08-29:
I think this needs scaling. Shouldn't PF be 1.0 or -0.55 or 0.58 etc?

dakaix, 2023-08-29:
Normally yes, and expressed as a unit of “p.f”.

I wonder though how the meter itself expresses it on the display… will be a few days before I can check that though!

wills106, 2023-08-29:
https://github.com/home-assistant/architecture/issues/469

Looks like I can either define "PERCENTAGE" or "None" so it works as a graph https://github.com/home-assistant/core/pull/85181/files

Which would you prefer?

dakaix, 2023-08-29:
Interestingly the Eastron meter’s Modbus documentation lists power factor as having no unit and as a float. That suggests Solis themselves may be translating this to a percentage.

https://eastroneurope.com/images/uploads/products/protocol/SDM120-MODBUS_Protocol.pdf

If this sensor is Solis specific then I would say go %, but if it’s generic then None would be better (as other vendors may not be scaling it up).

wills106, 2023-08-29:
I have changed it to None in [2023.08.7b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.7b3) and scaled it.

Can you let me know if the Graphing works correctly with None. If it does I'll update the Sofar plugin as well.

dakaix, 2023-08-29:
So good news, graphing now works... bad news, with "None" as the Unit it literally appends that to the UI!
[image]

wills106, 2023-09-01:
Should be corrected in [2023.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.1)

dakaix, 2023-09-01:
That’s back to displaying as a state change I’m afraid. 
[image]
[image]

wills106, 2023-09-01:
I have added in an extra state_class to force it back into a measurement [2023.09.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.2)

wills106, 2023-09-02:
Did you manage to test 2023.09.2 out?

Just wondering if I can close this down now?

dakaix, 2023-09-02:
Yes looking much better now! Thanks for jumping on this so quickly! :)

[image]

wills106, 2023-09-02:
I'll close this down then!

closed 2023-09-02 by wills106