wills106/homeassistant-solax-modbus #510: [BUG]

raenji-sk, 2023-08-28:
**Describe the bug**
Integration not working after upgrade to latest version. No data from inverter.
Worked fine on previous version 2023.3.x

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:342
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 10:24:42 AM (166 occurrences)
Last logged: 10:38:24 AM

General exception: unpack requires a buffer of 2 bytes
Modbus IO exception Modbus Error: [Input/Output] Unable to decode request

**Mandatory details**
1. Integration Version: 2023.08.06
2. HA Version: 2023.8.4
3. Inverter Brand: Solax X3 MIC G2
4. Plugin Used:
5. First 6 digits of Serial: MC206T
6. Connection Method: Modbus WiFi

**Detailed Error Log**
Home Assistant Core
2023-08-28 10:23:38.858 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-08-28 10:24:30.383 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX-x3-mic-g2: reading serial number from address 0x0 failed; other address may succeed

**Additional context**
Add any other context about the problem here.

raenji-sk, 2023-08-28:
Tried to remove the integration, restart HA, reinstall, no help. Still getting the same error

raenji-sk, 2023-08-28:
Reverted manually back to 2023.03.1 and all works fine.

wills106, 2023-08-28:
Looks like you had some bad Modbus data come through.
Did you try power cycling the Modbus adaptor?

raenji-sk, 2023-08-28:
No, but I can try sometime later when I decide to try to upgrade again. I was upgrading whole HA and thought it would be a good time to upgrade everything else while I was at it. But otherwise v2023.3.1 works just fine for me. So I'll probably stick with it untill an upgrade is forced by HA compatibility or similar issue.

wills106, 2023-09-02:
Can I close this issue down now?

raenji-sk, 2023-09-02:
@wills106 yes, for the moment this issue is irrelevant.

closed 2023-09-02 by wills106