wills106/homeassistant-solax-modbus #513: TIGO TSI Inverters

FloCodeRian, 2023-08-28:
**Describe the bug**
Hi there, Want to use your plugin with a TIGO ENERGY Hybrid Inverter like TSI-15K3D which is not that different I thought than those hybrid G3 from Qcells. 
Modus ist used by Tigo Energy CCA for getting inverter Reading in their app and Webportal. 
I can't get any readings out of the inverter by accessing over der modbus RTU via the USB to Modbus adapter shown in the wiki.
First was to try it without the CCA in parallel, but it doesn't get any better.
I would by pleased if anyone can help. 

**Mandatory details**
1. Integration Version: 2023.08.6
2. HA Version: 2023.8.4
3. Inverter Brand: TIGO ENERGY TSI-15K3D (Label TSS-3PS)
4. Plugin Used: Solax
5. First 6 digits of Serial: 6513000.... Inv; BMS 660000
6. Connection Method: Modbus RTU 

**Detailed Error Log**

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:5322 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 25. August 2023 um 23:53:56 (6 occurrences) 
Last logged: 20:02:40

unrecognized inverter type - serial number : unknown
unrecognized inverter type - serial number : H3151010H1xxxx
unrecognized inverter type - serial number : 3H5101011Hxxxx

**Additional context**
Add any other context about the problem here.

wills106, 2023-08-28:
Just added it into [2023.08.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.08.7b1) Looks like a Gen4 SolaX Hybrid

> unrecognized inverter type - serial number : H3151010H1xxxx

You'll want to use whichever method you used to get that Serial Number response.
If it works I can properly add it into the next stable release.

FloCodeRian, 2023-08-28:
Thank you for your help!
I’m very pleased about your competence! 

But can you help me by your task? 
Any steps or reference for the wiki?
Or meet with PN?

FloCodeRian, 2023-08-28:
However it worked out but I played with the inverter model in the setup menu after installing your recommended beta version. 

It is now working on most register I would say. There is still a problem with the serialnumber read. 
If you tell me a good way I can share which register are already working and which aren’t. 

Thank you very much in any case!

wills106, 2023-08-29:
Just normal screen shots is fine.

FloCodeRian, 2023-08-31:
[20230831-Solaxmodbus_Tigo.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/12490134/20230831-Solaxmodbus_Tigo.pdf)
that's all I can see so far.

I think the entity: Tigo Export Control User Limit isn't correct too because I have a 15kW inverter und 2500W is the maximum home assistant lets do me, but it works kind of expected though ... at the inverter I can change it from 0 to 30000W which doesn't make sense too but works.  missing battery information reasons maybe in communication with the inverter and could be a tigo problem

wills106, 2023-09-01:
> 2500W is the maximum

I didn't fully add your Inverter in until I knew it return things correctly.
Looks like it is just a rebranded Gen4 so I'll add the Inverter in correctly.

> 0 to 30000W

#437 gives an example why you get set it to such a large value.

wills106, 2023-09-01:
Should be corrected in [2023.09.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.1)

wills106, 2023-09-02:
Have you tested 2023.09.1 / 2023.09.2 out?

Just wondering if I can close this down now?

FloCodeRian, 2023-09-02:
I have updated and tested and it’s kind of working but not that much more than the beta I used earlier. See the picture below. Thank you for Your help!

wills106, 2023-09-02:
That mostly seems fine other than the PV Output. Do you have any PV Panels connected? Or are you using this as battery storage only?

FloCodeRian, 2023-09-02:
I'm using the system kind of normal with battery and some panel connected.
Still though the small PV output is because of the "zero feedin" setting from me. 
Which I'm going to change after I'm allowed to.

Yes I'm using all panels with optimizers and a tap connect to their CCA - since the refreshrate through the Tigo servers is buggy and very slow. I've connected it only to my homeassistant setup. Formerly it was connected in parallel to the RS485 bus.

I would also love to get those information about the optimizers as fast as your integration works. 
But there aren't any docs about the rs485 between the Tap and the CCA.

wills106, 2023-09-03:
I'm going to close this down as I can't make any improvements without further documentation.
If you are able to obtain any then raise a [Feature Request] and I can finish off the implementation.

closed 2023-09-03 by wills106

FloCodeRian, 2023-10-15:
Need help again,
Updated os to 11.0 and core 2023.10.3

And whole integration stopped working suddenly. 

Now telling me about unknown serialnumbers. Because didn't find them. 

Downgrade OS to 10.5 and core 2023.10.1 didn't help! 

Any further Ideas?

wills106, 2023-10-16:
> Unfortunately only the USB-Serial port changed. After those restarts during the update.
> 
> Can I fix this for ever, somehow?

You could try giving it the id of you usb adaptor. Although I don't know if the integration will work i that manner? I have no way of testing.

It will be something like:
/dev/serial/by-id/usb-xyz

If you have more than one usb serial adaptor you will have to try and establish which one is which.
If it does work with serial/by-id can you let me know and I can add it into the docs?

FloCodeRian, 2023-10-17:
Just replaced the /dev/ttyUSB0
With /dev/serial/by-id/usb-1a86_USB2.0-Ser_-if00-port0

And it's still working. :)