wills106/homeassistant-solax-modbus #525: Solis House Load - zero readings

bulletcom, 2023-09-04:
1. Integration Version: 2023.09.2
2. HA Version: 2023.8.4
3. Inverter Brand: Solis Hybrid 6K
4. Plugin Used: SolaX Inverter Modbus
5. First 6 digits of Serial: 603105
6. Connection Method: Elfin EW11 using this method: https://github.com/Jumpy07/Solis---SolisCloud-and-Home-Assistant

Issue:
Solax House Load is periodically reading zero. All other entities report correctly. Have restarted all components including inverter but still happening. 

[image]

Any ideas? 

SolisCloud readings are ok (albeit at 5 min intervals) so I don't think it's the inverter.

alienatedsec, 2023-09-04:
Interesting that I have similar readings but never paid any attention to that. I'm using waveshare so it's not the connection method. [image]

wills106, 2023-09-08:
Are actually getting errors when you get a zero reading?
If there isn't, I can't really do anything about it.

wills106, 2023-09-18:
Release [2023.09.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.5) won't address communication issues, but should prevent dashboard spikes if a `None` value was sent. But if the Inverter is genuinely sending 0W house load then there isn't much we can do about that as it's just reporting what the Inverter tells it.

wills106, 2023-09-23:
Closing down as no further response from original poster.

closed 2023-09-23 by wills106