wills106/homeassistant-solax-modbus #526: [BUG] Spikes in Battery Charging in Energy Dashboard

DidMyBest, 2023-09-04:
**Description**
Since 03rd of September 2023 I had twice a extreme Spike in HA Energy Dashboard for Solax Battery Charging. I could eliminate the Spike via HA Developer Tools --> Statistics (using the ramp icon). But at exactly the same time the spikes appeared an error message can be found in the log. See below for error log.
I did not change anything in the last few weeks in the hardware or software, so I am not sure, where this originates from.

**Mandatory details**
1. Integration Version:  2023.09.02
2. HA Version: Home Assistant 2023.8.4, Supervisor 2023.08.1, Operating System 10.5, Frontend 20230802.1 - latest
3. Inverter Brand: Solax 
4. Plugin Used: not sure what's meant here
5. First 6 digits of Serial: H34A08
6. Connection Method: ModBus

**Detailed Error Log**
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:288
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 3. September 2023 um 11:09:03 (14 occurrences)
Last logged: 02:26:11

SolaX: read failed at 0x4a: grid_import_total
SolaX: read failed at 0x4f: eps_frequency
SolaX: read failed at 0x50: today_yield
SolaX: read failed at 0x52: total_yield
SolaX: read failed at 0x54: lock_state

**Further Error Logs from my HA Installation regarding Solax**
Error 1: 
Logger: homeassistant.util.async_
Source: util/async_.py:166
First occurred: 1. September 2023 um 21:09:23 (1013223 occurrences)
Last logged: 11:09:24

Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 336: return self._client.read_holding_registers(address, count, **kwargs)
Detected blocking call to sleep inside the event loop. This is causing stability issues. Please report issue to the custom integration author for solax_modbus doing blocking calls at custom_components/solax_modbus/__init__.py, line 342: return self._client.read_input_registers(address, count, **kwargs)

Error 2:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:336
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 1. September 2023 um 21:09:28 (6246 occurrences)
Last logged: 11:09:08

Cleanup recv buffer before send: 0xaa 0x55 0xf 0xc8 0xff 0x1 0x0 0x0 0x1 0x1 0x0 0x1 0x64 0x3d 0x3
Cleanup recv buffer before send: 0xaa 0x55 0xf 0xc8 0xff 0x1 0x1 0x0 0x1 0x1 0x0 0x1 0x64 0x3e 0x3

Error 3:
Logger: homeassistant.helpers.template_entity
Source: helpers/template_entity.py:362
First occurred: 2. September 2023 um 00:01:14 (52 occurrences)
Last logged: 11:08:22

TemplateError('UndefinedError: 'None' has no attribute 'value'') while processing template 'Template<template=({% set ftd = state_attr("sensor.ebusd_vr71_mixermovement", "FTDesired").value | float %} {## Flowtempdesired ##} {{ ftd | round(1) }}) renders=8>' for attribute '_attr_native_value' in entity 'sensor.vaillant_mixer_flowtempdesired'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set cop1 = states('sensor.cop_aktuell') | float %} {% set pef1 = states('sensor.primaernergiefaktor') | float %} {{ (cop1/pef1*100) | round(1) }}' but no default was specified') while processing template 'Template<template=({% set cop1 = states('sensor.cop_aktuell') | float %} {% set pef1 = states('sensor.primaernergiefaktor') | float %} {{ (cop1/pef1*100) | round(1) }}) renders=76>' for attribute '_attr_native_value' in entity 'sensor.etas_own'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set sbiet = states('sensor.solax_battery_input_energy_today') | float %} {% set sboet = states('sensor.solax_battery_output_energy_today') | float %} {% set stie = states('sensor.solax_today_s_import_energy') | float %} {% set stee = states('sensor.solax_today_s_export_energy') | float %} {% set stse = states('sensor.solax_today_s_solar_energy') | float %} {{ stse+stie+sboet-sbiet-stee | round(1, default=0) }}' but no default was specified') while processing template 'Template<template=({% set sbiet = states('sensor.solax_battery_input_energy_today') | float %} {% set sboet = states('sensor.solax_battery_output_energy_today') | float %} {% set stie = states('sensor.solax_today_s_import_energy') | float %} {% set stee = states('sensor.solax_today_s_export_energy') | float %} {% set stse = states('sensor.solax_today_s_solar_energy') | float %} {{ stse+stie+sboet-sbiet-stee | round(1, default=0) }}) renders=2086>' for attribute '_attr_native_value' in entity 'sensor.energieverbrauch_taeglich'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set evt = states('sensor.energieverbrauch_taeglich') | float %} {% set stie1 = states('sensor.solax_today_s_import_energy') | float %} {% set pef = 2.5 if stie1 > evt else ((stie1/evt*2.5)+((evt-stie1)/evt)) if stie1 < evt else 0 | round(1) %} {{ pef | round(1) }}' but no default was specified') while processing template 'Template<template=({% set evt = states('sensor.energieverbrauch_taeglich') | float %} {% set stie1 = states('sensor.solax_today_s_import_energy') | float %} {% set pef = 2.5 if stie1 > evt else ((stie1/evt*2.5)+((evt-stie1)/evt)) if stie1 < evt else 0 | round(1) %} {{ pef | round(1) }}) renders=2280>' for attribute '_attr_native_value' in entity 'sensor.primaernergiefaktor'
TemplateError('ValueError: Template error: float got invalid input 'unavailable' when rendering template '{% set cop1 = states('sensor.cop_aktuell') | float %} {% set pef1 = states('sensor.primaernergiefaktor') | float %} {{ (cop1/pef1*100) | round(1) }}' but no default was specified') while processing template 'Template<template=({% set cop1 = states('sensor.cop_aktuell') | float %} {% set pef1 = states('sensor.primaernergiefaktor') | float %} {{ (cop1/pef1*100) | round(1) }}) renders=162>' for attribute '_attr_native_value' in entity 'sensor.etas_own'

Error 4:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:342
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 1. September 2023 um 21:23:29 (393 occurrences)
Last logged: 11:05:37

Cleanup recv buffer before send: 0xaa 0x55 0xf 0xc8 0xff 0x1 0x0 0x0 0x1 0x1 0x0 0x1 0x64 0x3d 0x3
Cleanup recv buffer before send: 0xaa 0x55 0xf 0xc8 0xff 0x1 0x1 0x0 0x1 0x1 0x0 0x1 0x64 0x3e 0x3

Error 5:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:288
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 3. September 2023 um 11:09:03 (14 occurrences)
Last logged: 02:26:11

SolaX: read failed at 0x4a: grid_import_total
SolaX: read failed at 0x4f: eps_frequency
SolaX: read failed at 0x50: today_yield
SolaX: read failed at 0x52: total_yield
SolaX: read failed at 0x54: lock_state

**Thanks**
Thanks for programming the Solax HA integration. It is great. Thanks for your reply and hopefully this bug can be mitigated.

infradom, 2023-09-05:
In my opinion, you had a temporary modbus serial communication problems. 
Furthermore, we recommend not to use a usb serial modbus dongle, but use an Ethernet based modbus adapter.
The way in which we handle serial RS485 modbus is not as async as it should be; afaik there are no plans to correct this as we recommend the use of Ethernet based adapters (see Wiki).
Are you sure Solax did not perform an online software upgrade at the time the problem occured?
You could check the history of the version numbers in HA

DidMyBest, 2023-09-05:
Thanks for your quick reply, @infradom. 
I will monitor, if the occurs more often. I installed the USB-RS485 Adapter, since it is listed officially in the Wiki (and it is cheaper and easier to install). If the problem occurs more often, I will switch to an ethernet-rs485 adapter.

wills106, 2023-09-18:
Release [2023.09.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.09.5) won't address communication issues, but should prevent dashboard spikes if a `None` value was sent.

DidMyBest, 2023-09-18:
A few days after my initial post, the serial adapter stopped working. I got the waveshare RS485 to Ethernet adapter and I can read all Solar data now. 
May I propose to take the serial adapter from the Wiki, so that others do not buy this adapter for this purpose anymore, if it is not recommended anyway?

wills106, 2023-09-23:
The USB adaptors still have there use, to establish if you have communications before buying something better. There are also different grades of USB-RS485 adaptors.

closed 2023-09-23 by wills106