wills106/homeassistant-solax-modbus #530: "Config flow could not be loaded"

jamespreedy, 2023-09-13:
**Describe the bug**

I'm having issues with they custom integration (bug https://github.com/wills106/homeassistant-solax-modbus/issues/528), so I wanted to try re-installing it.

I deleted my devices and rebooted. I then removed the integration using the HACS and rebooted Home Assistant.
I checked that the solax_modbus folder had been removed from the custom_components folder (it had).

I then rebooted for good luck, and then re-installed v2023.9.3 using HACS.
I rebooted after the install, and then tried to set the component up from Settings => Devices & Services => Add Integration => SolaX Inverter Modbus.

After selecting the integration, I am then presented with an error saying "Config flow could not be loaded" - I've added a screenshot below.

I've tried basic troubleshooting - restarting, reinstalling the integration, installing the integration manually rather than using HACS, even tried the latest beta - all with no change.

[image]

**Mandatory details**

1. Integration Version:2023.9.3
2. HA Version:2023.9.2
3. Inverter Brand:SolaX
4. Plugin Used:solax
5. First 6 digits of Serial: H34A10
6. Connection Method:Waveshare RS485 to Eth

**Detailed Error Log**
Nothing mentioning solax in the logs

Thanks in advance for your help!

jamespreedy, 2023-09-13:
@wills106 - I managed to find some logs to maybe help here. 🤞🏻

I get no errors on Home Assistant boot. But when I try to add the device as described above (Settings => Devices & Services => Add Integration => SolaX Inverter Modbus), I get the below errors in my logs.

Happy to get anything else you think might be useful - thanks!

```
2023-09-13 20:38:35.158 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solax_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 838, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 855, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little
2023-09-13 20:38:35.162 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration solax_modbus: Exception importing custom_components.solax_modbus.config_flow
2023-09-13 20:38:48.701 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solax_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 838, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 855, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little
2023-09-13 20:38:48.704 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading flow for integration solax_modbus: Exception importing custom_components.solax_modbus.config_flow
```

wills106, 2023-09-13:
> byteorder=Endian.Little, wordorder=Endian.Big,

You are still on Integration version 2023.09.3 with HA 2023.9.x by the looks. Have you tried the Integration 2023.09.4b2?

jamespreedy, 2023-09-13:
Hey wills106, yes, sorry, forgot to mention I tried both - below is with b2.

```
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solax_modbus/__init__.py:31 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 9:27:23 pm (1 occurrences) 
Last logged: 9:27:23 pm

Unexpected exception importing platform custom_components.solax_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 838, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 855, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

Logger: homeassistant.config_entries
Source: config_entries.py:864 
First occurred: 9:27:23 pm (1 occurrences) 
Last logged: 9:27:23 pm

Error occurred loading flow for integration solax_modbus: Exception importing custom_components.solax_modbus.config_flow

```

jamespreedy, 2023-09-14:
@wills106 - thanks for the update. Just tried b3, no change I'm afraid.

```

This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solax_modbus/__init__.py:31 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 7:35:46 am (1 occurrences) 
Last logged: 7:35:46 am

Unexpected exception importing platform custom_components.solax_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 838, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 855, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little

```

jamespreedy, 2023-09-14:
@wills106 ... same with b4 I'm afraid.

```
This error originated from a custom integration.

Logger: homeassistant.loader
Source: custom_components/solax_modbus/__init__.py:31 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 9:40:58 am (1 occurrences) 
Last logged: 9:40:58 am

Unexpected exception importing platform custom_components.solax_modbus.config_flow
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 838, in get_platform
    cache[full_name] = self._import_platform(platform_name)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 855, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1126, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/__init__.py", line 31, in <module>
    from pymodbus.payload import BinaryPayloadBuilder, BinaryPayloadDecoder, Endian
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 28, in <module>
    class BinaryPayloadBuilder:
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 42, in BinaryPayloadBuilder
    self, payload=None, byteorder=Endian.Little, wordorder=Endian.Big, repack=False
                                  ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/enum.py", line 784, in __getattr__
    raise AttributeError(name) from None
AttributeError: Little
```

jamespreedy, 2023-09-15:
@wills106  - issues resolved but updating to b5, I'm back up and running!
Thank you SO MUCH for all your help again. 🍺

wills106, 2023-09-15:
I'll close this down then.

closed 2023-09-15 by wills106