wills106/homeassistant-solax-modbus #537: [Possible BUG]

kimb1, 2023-09-19:
Solis 5kW, Waveshare RS485 to Wifi using Modbus TCP - Modbus RTU gateway mode.

Seems generally to work - a lot of addresses read fine.  I can aparently read addresses with other modbus tools separately.

Integration gets down to "battery_charge_direction"  and dies with an error reading modbus data.

`2023-09-19 23:38:32.901 ERROR (MainThread) [pymodbus.logging] General exception: unpack requires a buffer of 2 bytes
2023-09-19 23:38:32.904 ERROR (MainThread) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
2023-09-19 23:38:32.907 ERROR (MainThread) [custom_components.solax_modbus] Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 436, in read_modbus_data
    res = self.read_modbus_registers_all()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 545, in read_modbus_registers_all
    self.data[descr.key] = descr.value_function(0, descr, self.data )
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/const.py", line 213, in value_function_battery_output_solis
    inout = datadict["battery_charge_direction"]
            ~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'battery_charge_direction'`

Home Assistant 2023.9.2 Supervisor 2023.09.2 Operating System 10.5
HomeAssistant-Solax-Modbus version": "2023.09.5

I've tried commenting that entry, and it gets stuck on the next one.

I'm wondering if it's possibly something like a temrination resistor giving bus errors, but the error is perfectly repeatable on this parameter - never any other.

Apologies if I've missed something trivial!

kimb1, 2023-10-03:
Plot thickens - I can communicate with the inverter using the python package pyModbusTCP   but not pymodbus   in TCP mode

The test script below will read three registers using "c1" but not "c2".    I can repeat this on the HomeAssistant instance and on a plain Debian box.     

I'll keep at it - I'm _sure_ I'm doing something stupid here.

`from pyModbusTCP.client import ModbusClient
from pymodbus.client import ModbusTcpClient

# init modbus client
c1 = ModbusClient(host='192.168.3.107', port=502, debug=True, auto_open=True)
c2 = ModbusTcpClient(host='192.168.3.107', port=502, debug=True, auto_open=True)

# if success display registers
print ("trying v1")
regs_l = c1.read_holding_registers(43006, 3)
if regs_l:
    print('reg ad #0 to 9: %s' % regs_l)
else:
    print('unable to read registers via c1')

print ("trying v2")
regs_2 = c2.read_holding_registers(43006, 3)
# if success display registers
if regs_2:
    print('reg ad #0 to 9: %s' % regs_2)
else:
    print('unable to read registers via c2')
~ #
`

alienatedsec, 2023-10-03:
https://github.com/wills106/homeassistant-solax-modbus/wiki/Compatible-RS485-Adaptors#non-working--troublesome-rs485-adaptors

```
[Waveshare RS485 to WIFI/ETH](https://www.waveshare.com/wiki/RS485_TO_WIFI/ETH)
This RS485 adaptor is very problematic:

There seems to be delayed communications,
Struggles to read more than 7 registers at a time,
Modbus Error: [Input/Output] Unable to decode request,
No ability to set "Auto query storage type" for Sofar Inverters.
```

kimb1, 2023-10-04:
Thank you - sorry, I'd missed that entry.   I got drawn in by other GitHub projects which seemed to use it successfully.

I now have it working with the Waveshare device in Modbus RTU over TCP mode, with the Waveshare set to "transparent".   So the Modbus protocol is now in python not in the Waveshare.    
Will monitor and see how it goes, but possibly put together a different device.
Thanks!

closed 2023-10-09 by kimb1

alienatedsec, 2023-10-09:
@kimb1 can you document/describe what you did to make it work?

kimb1, 2023-10-10:
As above:   I have a configuration which has been working for several days apparently successfully

- Solis RHI-5K-48ES-5G
- Waveshare RS485 to WIFI/ETH set in "Transparent "mode - see section 4.1.1 in [](https://files.waveshare.com/upload/6/61/RS485-TO-WIFI-ETH-User-Manual-EN.pdf)
- Waveshare powered from Inverter 5v on datalogger plug
- Serial port set to match inverter (9600, 8,n,1)
- Home Assistant 2023.10.1
- SolaX Inverter Modbus 2023.09.7

SolaX Inverter Modbus configured:
- Interface: TCP
- Inverter Type: Solis
- Modbus address 1
- Polling 15s
- IP address to match Waveshare
- Port as configured in Waveshare (I used 502, which isn't the default there but is normal for Modbus
**- Modbus RTU over TCP**

So the key is not using the Waveshare to do Modbus/TCP to Modbus over serial conversion itself.    
The Modbus protocol is now in Python in HomeAssistant and the TCP connection and Waveshare are a "dumb pipe for bytes"    This seems to avoid the issues in the FAQ reading more than 7 registers at a time.

Because the Waveshare has no awareness of open or completed Modbus requests I think it would be inadvisable to have multiple clients connect to the Waveshare (as is apparently possible in some other projects to have the Solis datalogger active to the Solis cloud simultaneously with HomeAssistant).   But in my setup I don't need that, only HA to the Solis.