wills106/homeassistant-solax-modbus #539: [BUG] Manual mode with force charge / discharge not working

alibasta, 2023-09-20:
**Describe the bug**
I try to write an automation to force charge the battery when the Tibber energy prices are low and the battery is nearly empty. Currently I try to change manually the following in the Solax HA device settings:

- **Charger Use mode** to **Manual Mode**
[image]

- **Manual Mode Control** to **On**
[image]

- **Manual Mode Select** to **Force Charge** (also tried force discharge)
[image]

In the HA Logfile I can see that the Charger Use mode switches back to the original value and Battery charging / discharging is not working.

**Mandatory details**
1. Integration Version: 2023.02.1
2. HA Version: 2023.8.3
3. Inverter Brand: Solax X3 Hybrid G4
4. Plugin Used:
5. First 6 digits of Serial:
6. Connection Method:

wills106, 2023-09-20:
Like infradom mentions, if your Inverter is indead "locked" check https://github.com/wills106/homeassistant-solax-modbus#faq Point 3

alibasta, 2023-09-21:
Thanks guys. It was my fault. The inverter was not unlocked yet! 

"Modbus Power Control (remotecontrol)" looks very exciting and like made for the intended purpose. Do you use the remotecontrol function to charge the battery in winter via a dynamic power tariff?

infradom, 2023-09-22:
> Do you use the remotecontrol function to charge the battery in winter via a dynamic power tariff?

Yes, I  use a dynamic power tariff and I indeed use it in the winter to charge the batteries when prices are low. In summer, spring, I use it to discharge batteries when prices are high. I also switch to the emulated 'feedin priority' when prices are intermediate.

wills106, 2023-09-24:
Closing down as initial issue resolved.

closed 2023-09-24 by wills106