wills106/homeassistant-solax-modbus #544: [BUG] Battery heating on Solax X3 does not work

Dehumanizer77, 2023-09-27:
**Describe the bug**
Cannot control battery heating (when switched from disabled to enabled, it turns back to disabled after few seconds) and the battery heating sensor shows always "disabled" even though heating is enabled on the device.

**Mandatory details**
1. Integration Version: 2023.09.6
2. HA Version: 2023.9.3
3. Inverter Brand: Solax X3 Hybrid G4
4. Plugin Used: ?
5. First 6 digits of Serial: H34A15
6. Connection Method: Pocket Wifi 3.0, Modbus TCP

**Detailed Error Log**
there are no errors in the log

Dehumanizer77, 2023-10-16:
only the heating setting

cewbdex, 2023-10-17:
Did you try the advanced unlock first?

Dehumanizer77, 2023-10-17:
yes I did, it didn't help, every other setting works, just heating shows always disabled (even though it's actually enabled) and cannot be set

cewbdex, 2023-10-17:
Then I don't know. Which firmware versions do you have on your inverter and pocketwifi?

Dehumanizer77, 2023-10-17:
where can I see the firmware versions?

Dehumanizer77, 2023-10-17:
master 27
manager 26
pocketwifi 06.04

Dehumanizer77, 2023-10-17:
I'll probably wait until I get my already ordered waveshare tcp/modbus converter and connect it to the modbus port to see if that will help. I want to get rid of the pocketwifi anyway and switch it for pocketlan

cewbdex, 2023-10-17:
You can try to update the pocketwifi fw at least, it takes a minute - https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/145866811000

I am running it with no issues at all, but I am not using battery heating.

Dehumanizer77, 2023-10-17:
Ok I will try it tomorrow and share the findings here.

wills106, 2023-10-17:
I don't think Battery Heating is correct in the Integration. I think earlier docs are wrong.

I need to relook at the Heating section, but it probably won't be until the weekend though.

wills106, 2023-10-17:
If you set one of the times to 00:01 on the Inverter / Cloud does it display as 01:00 in the Integration?
Or set 01:00 on the Inverter / Cloud and the Integration displays 00:01?

Dehumanizer77, 2023-10-17:
Well it doesn't seem to change. Start time 1 is always 00:00, Start time 2 is always 01:00 and both End time 1 and End time 2 are always 00:00, no matter what I set on the inverter. When I try to change it in the integration, it always changes back to these values after a few seconds.

Dehumanizer77, 2023-10-25:
@wills106 it seems some other settings behave the same way. I now wanted to set solax_backup_discharge_min_soc and again, I change it and a few seconds after it switches back to the previous value. When I set it using solaxcloud app, it changes correctly and the value of solax_backup_discharge_min_soc changes in the HA integration accordingly (so it shows the value I have set in the app). So, it basically acts just as a read-only sensor.
I suppose there will be more settings acting like this, but I didn't have a chance to check anything else yet.

PatrikTrestik, 2023-10-25:
@wills106 I have Solax G4 and I'm programer. I'm going to play with this as I will soon require battery heating too.

BTW: Not all batteries have this function. It depends on battery type not only inverter type. Test first through cloud app if you are able to set it up and it works.

wills106, 2023-10-25:
Yea I'm only aware of the T3.0 batteries supporting this.

> I don't think Battery Heating is correct in the Integration. I think earlier docs are wrong.
> 
> I need to relook at the Heating section, but it probably won't be until the weekend though.

Still need to double check the registers as I do believe they are wrong in the Integration at the moment.

PatrikTrestik, 2023-10-25:
> Yea I'm only aware of the T3.0 batteries supporting this.
>
>
> Still need to double check the registers as I do believe they are wrong in the Integration at the moment.

I'm just now playing with Battery Heating registers. Might be I'll come with PR soon.

Dehumanizer77, 2023-10-25:
> @wills106 I have Solax G4 and I'm programer. I'm going to play with this as I will soon require battery heating too.
> 
> BTW: Not all batteries have this function. It depends on battery type not only inverter type. Test first through cloud app if you are able to set it up and it works.

In my case I know my batteries and inverter do support heating, as it was a requirement when ordering, I have batteries in garage without heating. And it works through the solax app.

PatrikTrestik, 2023-10-25:
> In my case I know my batteries and inverter do support heating, as it was a requirement when ordering, I have batteries in garage without heating. And it works through the solax app.

Happy you. I just now found that my battery does not support heating even thou it is installed in garage without heating. I have to talk to installation company (and later on install external heating :) ).

Result of my fiding is that I'm at no help with testing and fixing this issue. Without working device I have no chance. Sorry.

PatrikTrestik, 2023-11-02:
> @PatrikTrestik do you think you could instead take a look at the other entities which seem to misbehave in the same way? I think it's the same problem so we might end up fixing the battery heating setting as well. As I said, for example solax_backup_discharge_min_soc also behaves the same way and you should be able to debug this one.

Sure. On Monday I have some time so I can try to find what is wrong.

PatrikTrestik, 2023-11-06:
> solax_backup_discharge_min_soc

I tested mentioned "solax_backup_discharge_min_soc" and it works well for me. Write is successfull and it keeps its value. I can confirm the value through SolaX Claud.

At same time I can confirm that "battery_heating" is reverted to Disabled after each write. I also don't see this option in SolaX Claud settings.

From this I guess that option has to be technicaly supported by Inverter. I don't have battery with heating option, that is why it does not work. 
Might be your invertor does not support backup mode or is in state where it is disabled. 

@Dehumanizer77 do you see option in SolaX Claud? Let me know if there is other parameter which you suspect, I'll test.

Dehumanizer77, 2023-11-06:
@PatrikTrestik That's really strange. I can confirm I see the option in Solax Cloud, see the screenshot:
[image]

But when I change it through number.solax_backup_discharge_min_soc it reverts back after a few seconds.

Dehumanizer77, 2023-11-06:
Also, if  change it in Solax Cloud, it works as expected.

Dehumanizer77, 2023-11-06:
@PatrikTrestik One more thing, do you have it connected via PocketWifi and TCP modbus or through the COM port and RS485/TCP converter?

wills106, 2023-11-06:
@Dehumanizer77 do you know what Firmware is on your PocketWiFi?
There is a post somewhere either on here or one of the FaceBook Groups about needing an update to support some newer features.

I don't know the latest version as I don't have a PocketWiFi, but back in June they were on version 3.009.03 You might be on a version that predates the battery heating registers. So might not be passing them on over Modbus?
The cloud connection uses a different connection method.

PatrikTrestik, 2023-11-06:
> @PatrikTrestik One more thing, do you have it connected via PocketWifi and TCP modbus or through the COM port and RS485/TCP converter?

I have PocketWifi with TCP modbus.
Inverter G4 latest firmware 1.33/1.31.
Solax integration 2023.10.1

It is strenge.

Dehumanizer77, 2023-11-06:
@wills106 My version is 3.001.02, I might try to upgrade. But I want to change it to the connection via the COM RS485 connection anyway.
Unrelated question, since the last integration upgrade, sensor.solax_firmware_version_inverter_master and sensor.solax_firmware_version_manager entities are unavailable, is this a known bug? I know I have seen the inverter versions in these entities before. 

@PatrikTrestik and what's your version of PocketWifi firmware?

PatrikTrestik, 2023-11-06:
> @wills106 My version is 3.001.02, I might try to upgrade. But I want to change it to the connection via the COM RS485 connection anyway. Unrelated question, since the last integration upgrade, [...]

3.001.02

wills106, 2023-11-06:
> Unrelated question, since the last integration upgrade, sensor.solax_firmware_version_inverter_master and sensor.solax_firmware_version_manager entities are unavailable, is this a known bug?

They got changed, but wasn't properly communicated in the changelog https://github.com/wills106/homeassistant-solax-modbus/issues/570

Dehumanizer77, 2023-11-06:
Ok it seems I have older version of the inverter firmware, I will upgrade and see if it will help. 
BTW @wills106 both major versions show 0 for me, even though it should be 1...

cewbdex, 2023-11-06:
> 3.001.02

This is not a correct version number, all versions show this. This is a version number:
[image]
In my case 3.010.16, the real latest version, you should upgrade to this version, as I told you before, before any further troubleshooting.

Dehumanizer77, 2023-11-06:
You are right, I totally forgot, I thought I have upgraded it. 
I will upgrade both pocketwifi and inverter firmware and I'll let you know if it helped.

Dehumanizer77, 2023-11-06:
OK, I know why I didn't upgrade. I haven't been able to find a pocketwifi firmware. There is no firmware at https://www.solaxpower.com/downloads/ or anywhere else. If anyone has a link for pocketwifi firmware, please let me know.

cewbdex, 2023-11-06:
I have provided you a link already...

Dehumanizer77, 2023-11-08:
Ok, I have upgraded the pocketwifi to the latest version @cewbdex has provided, it did not help. 
However, I am quite confused about the inverter versions. This is what I see on the inverter itself: 

[image]
[image]

and in the integration I see this:

[image]

while the latest firmware is supposedly 1.33/1.31

Can anyone explain it to me?

cewbdex, 2023-11-08:
Solax for some reason does not change the firmware version value of their product, you can google that. The one integration provides are correct (they are as internal codes inside the inverter). In the parent folder of the one I provided are even newer firmwares, with updated regional regulations.

Dehumanizer77, 2023-11-09:
Thank you for the explanation.
In the meantime, I have received the RS485 to TCP adaptor and successfully connected the inverter through the modbus (COM) port on the inverter, but it didn't fix either the battery heating issue nor the min SOC issue, so it's definitelly not a problem with pocketwifi. I will wait until I'll upgrade the inverter (I can't right now, as they say with X3 G4 you have to have some minimum voltage on panels so you should do it on a sunny day and it's still periods of sun and clouds here for the last few days so I don't want to risk bricking the invertor) and if the upgrade won't fix it, then I really don't know.

So just to sumarize the issue, both sensor.solax_battery_heating and select.solax_battery_heating show "Disabled" in the integration even though the heating is actually enabled (as per solaxcloud), when I check enabled, it goes back to disabled after few seconds, but the solax_backup_discharge_min_soc shows correct value (which is set on the inverter) and it changes when I change the value through the cloud, it just cannot be changed through the integration.

cewbdex, 2023-11-09:
I have two versions of the modbus protocol document and the heating entities are not present in the older one (showed as reserved), they could be added in a latter release, but then I don't know how the SolaX cloud manages the information, capture of communication would be nice if nothing else helps.

Dehumanizer77, 2023-11-09:
@cewbdex you mean the communication between the inverter and cloud server? I can capture something if you tell me what exactly are you looking for. Like try to interact with the heating settings and capture the communication?
I have just done that and I have the captured packets, but I'm not comfortable with pasting it here since I don't know what exactly does it include. I can see the device ID there in cleartext among some other data, so at least it seems it's not encrypted.

wills106, 2023-11-10:
I'm not sure if the captured data will provide anything useful?
The cloud communicates to the Inverter over MQTT.

So I wouldn't expect anything to match up to be honest.

cewbdex, 2023-11-10:
That was just a blind shot, then maybe the modbus battery heating control might not be implemented in his fw version?

Dehumanizer77, 2023-11-10:
Let's wait until I upgrade the firmware, I'll let you know.

wills106, 2023-11-10:
It might be worth reaching out on the solaxinstallation facebook group to see if anyone is controlling the heating with this Integration?

I can't find any obvious feedback in the discussions page on here or in my emails.
So I honestly don't know if it works for anyone or not. I don't have a way to test and I can only go off the docs.

Dehumanizer77, 2023-11-16:
Ok so finally we have a sunny day here so I have upgraded the inverter to the very latest version 1.34/1.33 and everything started to work. Heating setting works and SOC setting works as well. So if anyone had the same problem in the future, just upgrade the inverter firmware.
Sorry for keeping you busy with this issue.

closed 2023-11-16 by Dehumanizer77