wills106/homeassistant-solax-modbus #555: [BUG]

glekkes, 2023-10-11:
**Describe the bug**
Unable to make a tcp modbus connection with S2-WL-ST datalogger.

**Mandatory details**
1. Integration Version: 2023.09.7
2. HA Version: 2023.10.0
3. Inverter Brand: Solis
4. Plugin Used: solis
5. First 6 digits of Serial: Inverter: 103103 / Datalogger: 7A1232
6. Connection Method: TCP Modbus

**Detailed Error Log**

See attached
[home-assistant_solax_modbus_2023-10-11T14-13-19.862Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/12870955/home-assistant_solax_modbus_2023-10-11T14-13-19.862Z.log)

**Additional context**
I use a wired connection through the S2-WL-ST datalogger from Solis. The connection seems to be refused from the looks of the logs.

alienatedsec, 2023-10-11:
The baud rate for this datalogger is 9600. I could see 19200 in the logs.

glekkes, 2023-10-11:
Thank you for that! But to my knowledge, that's only applicable when you use a serial connection.
Anyway, I changed it, but the issue remained the same.

alienatedsec, 2023-10-11:
> Thank you for that! But to my knowledge, that's only applicable when you use a serial connection. Anyway, I changed it, but the issue remained the same.

I misread this line
```
2023-10-11 16:13:19.412 DEBUG (MainThread) [custom_components.solax_modbus] solax modbushub creation with interface tcp baudrate (only for serial): 19200
```

What is the firmware version of datalogger? is the port `502` open when trying `nmap`?

glekkes, 2023-10-17:
Sorry for the delay in my answer. The port is not open. But I can't seem to find any open on how to open the port. FW version is 10010106.

alienatedsec, 2023-10-17:
> FW version is 10010106.

- @glekkes  `10010119` on mine - screenshot below

[image]

- NMAP output
[image]

glekkes, 2023-10-18:
Mmm, that might be the cause. Do you know by any chance how I might upgrade the fw? It seems there is no documentation available to achieve this.

alienatedsec, 2023-10-18:
@glekkes try raising a ticket with Solis directly. You need to give them your data logger details like model and serial number. 

One thing to remember; as soon you start using the S2-WL-ST data logger with ANY integration in HA, it will stop reporting to the SolisCloud.

glekkes, 2023-10-18:
Raised the ticket, very fast response! They upgraded  the FW, problem solved. 👍 Thank you for your support! (Strangely, they do not seem to be aware of thus functionality)

alienatedsec, 2023-10-18:
@glekkes thats great to hear

What is your firmware now?

glekkes, 2023-10-18:
It's 10010117 now

glekkes, 2023-10-18:
Closed, fixed by updating Firmware of datalogger

closed 2023-10-18 by glekkes