wills106/homeassistant-solax-modbus #559: Solax lan pocket

telheiras024, 2023-10-15:
Hello, I'm trying to integrate my solax with HA but withou success.

Only appears 4 identities and with the following error at HA logs - 

This error originated from a custom integration.

Logger: pymodbus.logging
Source: custom_components/solax_modbus/__init__.py:344
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: October 14, 2023 at 18:06:08 (6 occurrences)
Last logged: 01:28:13

Connection to (192.168.68.132, 502) failed: [Errno 111] Connection refused

Please, can you help me?

Thank you

wills106, 2023-10-16:
The PocketLAN doesn't provide a Modbus connection only the PocketWiFi, although I suggest using a device like the Waveshare as they provide a more reliable connection.

Check the Wiki for the correct version.

closed 2023-10-16 by wills106

telheiras024, 2023-10-16:
> The PocketLAN doesn't provide a Modbus connection only the PocketWiFi, although I suggest using a device like the Waveshare as they provide a more reliable connection.
> 
> Check the Wiki for the correct version.

Thank you for your quick answer.
Regarding your comment, to check the wiki, I have tried to search pocketLAN and Waveshare but didn't find nothing. what should I check? Or with PocketLAN I don't have any kind of integration with HomeAssistant?
One again, THANK you!