wills106/homeassistant-solax-modbus #566: [Question]

Kasperfcb, 2023-10-18:
I'm using solax in my home assistant. I have growatt sph3600 inverter, but I can't find the sensor where I can see how much I import from grid. Anyone can help me?

Kasperfcb, 2023-10-19:
> Check disabled entities.

I have checked there but nothing is hidden or disabled

MichaelBaker2020, 2023-10-21:
Hello I'm very new to HA and I am trying to set up the Solax integration and I have similar issues. Solax integration with Solax  X1-Hybrid-G4 I have the correct IP yet I have only 4 entities and can't find a way to unlock them? Help

<br class="Apple-interchange-newline" style="caret-color: rgb(0, 0, 0); color: rgb(0, 0, 0); font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration: none;">

wills106, 2023-10-21:
@MichaelBaker2020 you don't have a working Modbus connection, how are you trying to connect?

wills106, 2023-10-22:
@Kasperfcb Can't you see "Today's Grid Import" and "Total Grid Import"?

Kasperfcb, 2023-10-22:
> @Kasperfcb Can't you see "Today's Grid Import" and "Total Grid Import"?

@wills106 Yes, I can see them, but I don't have any that are updated every 15 seconds

MichaelBaker2020, 2023-10-22:
> @MichaelBaker2020 you don't have a working Modbus connection, how are you trying to connect?

Thanks for this I was coming to the same conclusion. I was trying the Solax WiFi route. I will make a hardware version. Then try again. Thanks again

PatrikTrestik, 2023-10-25:
Please note that only new Solax WiFi dongle (black one) has open and working Modbus TCP. 
There are some ways to enable it on other dongle, but you have to search and try.

wills106, 2023-10-28:
> > @Kasperfcb Can't you see "Today's Grid Import" and "Total Grid Import"?
> 
> @wills106 Yes, I can see them, but I don't have any that are updated every 15 seconds

Do you have either CT Clamps or a Modbus Power Meter connected to your Inverter?
If you don't your Inverter won't know how much you are Importing / Exporting from the Grid.

Do they always remain 0?