wills106/homeassistant-solax-modbus #570: Entities - Unavailable ARM/DSP Firmware version

matomatusov, 2023-10-21:
**Mandatory details**
1.Integration Version: 2023.10.1
2. HA Version: 2023.10.2
3. Inverter Brand: Solax X3 Gen4 10kW
4. Plugin Used:
5. First 6 digits of Serial: H34A10
6. Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

Solax
After the update to 2023.10.1 it stopped working firmware version.

**Detailed Error Log**
This entity is no longer being provided by the solax_modbus integration. If the entity is no longer in use, delete it in settings.

[image]

cewbdex, 2023-10-21:
This is no bug, thing were changed under the hood, so it matches the reality - master for G3 and older and ARM for G4. Remove entities and add the correct new ones:
[image]

matomatusov, 2023-10-21:
Thank you for the information, it works.

closed 2023-10-21 by matomatusov

wills106, 2023-10-22:
I forgot to add it to the change log on 2023.10.1