wills106/homeassistant-solax-modbus #571: Solax: Not discharging from the mains

otava5, 2023-10-22:
Hello. I have a general problem that is not related to the add-on and I am asking for some advice….
Charging from the mains in self-consumption mode:
When I have it set to "disable" and I set the charging time to 00:00-23:59, it does not drain my battery for consumption.
When setting 00:00-00:00 and "disable", everything is fine
Secondary charging from mains works fine.
Solax fw 31/33

Thanks for the advice

alienatedsec, 2023-10-22:
@otava5 No matter which inverter brand, you cannot charge your battery and discharge in self-consumption mode at the same time. 

When you set 00:00 - 00:00 you disabled charging from the grid which allowed discharging in self-consumption. 

Don't set the charging window to 00:00 - 23:59 as you instructed to charge from midnight until a minute before midnight the same day, so 23 hours and 59 minutes. 

Try 22:59 — 23:59 which is an hour.

otava5, 2023-10-22:
Attention, I write that I have the battery charging on "**DISABLE**" and it does not discharge the battery