wills106/homeassistant-solax-modbus #573: Waveshare RS485 TO ETH (B)

matomatusov, 2023-10-22:
I have a question for Waveshare RS485 TO ETH (B). After an internet outage, the given Waveshare RS485 TO ETH (B) loses communication between HA and Waveshare RS485 TO ETH (B).
You don't know why?
When there is a power failure, everything works.

THANK you!

cewbdex, 2023-10-23:
Could you describe your network setup a bit? What do you mean by internet outage? Upstream WAN error or issue on your local equipment?

matomatusov, 2023-10-23:
Waveshare RS485 TO ETH (B) setup is according to https://www.waveshare.com/wiki/RS485_TO_ETH_(B). 
By the Internet outage, I mean that the Internet is not coming from the provider. HA is installed on a Synology NAS.

cewbdex, 2023-10-25:
I don't have a personal experience with those devices, but could you provide a map of your network and which network devices do you use? Is there anything in the log (both sections)?

alienatedsec, 2023-10-25:
It feels like the same device that has the Waveshare ethernet connected is going down at the same time as the device that manages the internet connection.

matomatusov, 2023-10-25:
It seems to me that a device that should receive modbus communication does not receive HA.
Or the device Waveshare that should send modbus communication does not send. After restarting the device Waveshare everything will start working. I wonder if it could be a problem with the device Waveshare.
@cewbdex What network map do you mean? The network is built on MikroTik (MikroTik Cloud Router Switch CRS326, MikroTik CSS326-24G-2S+RM and 3xAP)

cewbdex, 2023-10-25:
Maybe could you also show the modbus adapter configuration? Does the disconnect happen after the internet goes down or after it restores?

PatrikTrestik, 2023-10-25:
Hi. 
Deeper knowledge of your LAN setup will be required.
In general WAN connection has nothing to do with communication between HA and Waweshare. If it does, there is some misconfiguration.

Please answer following:
- IP of Waweshare (or screenshot of setup page)
- IP of HA (local LAN IP)
- if possible output of "ipconfig" from any LAN connected Win computer.
- Are you able to simulate it? For example switch off network provider box.

matomatusov, 2023-10-25:
@cewbdex I am sending a preview of the adapter configuration. There is a dropout when connecting to the internet.
[image]

matomatusov, 2023-10-25:
@ PatrikTrestik

[image]
[image]

yes, I can simulate by turning off the internet from the provider.

PatrikTrestik, 2023-10-25:
@matomatusov It looks like this issue has nothing to do with Solax integration. I suspect some HA/LAN configuration problem.

Drop me message on "patrikt (at) volny.cz" so we can solve it offline.

cewbdex, 2023-10-26:
Probably not related, but multihost is recommended to be off, as [in the installation instructions](https://github.com/wills106/homeassistant-solax-modbus/wiki/Installation-Notes). Also I don't see the reason to make IPs less readable, as that could help to diagnose the problem too, those are your private IPs, so it's pointless, the only private entry is the MAC and you left it to the public. Is the IP you statically set outside of router's DHCP range?

matomatusov, 2023-10-26:
@cewbdex
Multihost is recommended to be on, because TiGO CCA is connected to it. The next host. The IP address is in the DHCP range (192.168.11.150).

cewbdex, 2023-10-26:
Well, the address should not be in the DHCP range, if the static setting I see sets static IP address of the device. Do you really have two modbus loggers attached to the inverter?

matomatusov, 2023-10-26:
Two Inverter and Tigo CCA devices are connected to the Waveshare device.

closed 2023-11-02 by matomatusov