wills106/homeassistant-solax-modbus #581: [Question] Use this Addon as separate docker container

aburgstah, 2023-10-26:
Hi I'm new to HA and all this things...
I'm running the HA as a docker (there are now addons available) 
So my question is, can I use this addon in a separate docker container? 
If so - How?
Thanks a lot!

wills106, 2023-10-26:
This isn't an add-on, it's a custom integration. So can either be manually installed by downloading a release or ideally through HACS.

So it will run in the HA container in Docker

wills106, 2023-10-28:
Going to close down as it's a question not an issue.

https://github.com/wills106/homeassistant-solax-modbus#installation

If you are still unsure how to install then start a discussion https://github.com/wills106/homeassistant-solax-modbus/discussions

closed 2023-10-28 by wills106