wills106/homeassistant-solax-modbus #582: [Question] Chrg&Dischrg Period Time

matomatusov, 2023-10-28:
Mandatory details:
1.Integration Version: 2023.11.1
2. HA Version: 2023.10.3
3. Inverter Brand: Solax X3 Gen4 10kW
4. Plugin Used: Solax Modbus
5. First 6 digits of Serial: H34A10
6. Connection Method: ModBus TCP using WaveShare RS485 TO ETH (B)

Why can't the step be set after minutes? In the SolaX Cloud app, this is set after minutes.
When you set 7:59 Charger End Time it will not appear in the SolaX integration.

[image]

wills106, 2023-10-28:
Because it's coded as a select before the datetime and time entities for an Integration were possible.
To prevent the select from having 1440 possible selections it's coded to mostly only work on 15min intervals.

I haven't been able to successfully get time entities working yet.

It's listed under the Todo List V2 https://github.com/wills106/homeassistant-solax-modbus/discussions/200#discussioncomment-5891065

matomatusov, 2023-10-28:
Thank you for the explanation.

closed 2023-10-28 by matomatusov

wills106, 2023-10-28:
If there is any successful progress it will be under the time branch https://github.com/wills106/homeassistant-solax-modbus/tree/time before a Beta goes out. I have put my work so far on that branch incase anyone knows what's missing to make it work.

matomatusov, 2023-10-28:
Then I can test when it is available.