wills106/homeassistant-solax-modbus #585: Battery Charge Upper SOC function does not work [BUG]

bluedevil500, 2023-10-30:
**Describe the bug**
The setting for the sensor mentioned above cannot be set via the integration. It works via the SolaX app. Maybe the address of the sensor to be saved is wrong?

**Mandatory details**
1. Integration Version: 2023.10.1
2. HA Version: 2023.10.5
3. Inverter Brand: SolaX Hybrid X3 G4.2
4. Plugin Used: SolaX Modbus
5. First 6 digits of Serial: H34A12
6. Connection Method: Modbus TCP

no log ...

wills106, 2023-11-03:
Please provide a log or something. I can't do much with "Does not work"

closed 2023-11-03 by wills106

bluedevil500, 2023-11-05:
unfortunately i cannot find a protocol for this error in ha

cewbdex, 2023-11-06:
You have described it in very badly, sensors are read, not set. You should try better next time, you told us nothing, no solutions you have tried, if you did any. We are not prophets to troubleshoot your problem.