wills106/homeassistant-solax-modbus #600: [Bug]: unrecognized inverter type - serial number : H4492AJ666XXXX

Joerivanrokeghem, 2023-11-07:
### Describe the bug

only shows up 4 entities, in debug I get the following error
2023-11-07 17:07:47.909 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : H4492AJ666XXXX

### Integration Version

2023.11.1

### Homeassistant core version

2023.11.1

### Inverter brand

Solax Power

### Plugin used

Solax

### First 6 digits of Serial

H4492A

### Inverter firmware versions

unable to find

### Connection Method

solax pocket wifi

### Dongle firmware

10.16

### I have done my own research

- [X] Yes, I did

wills106, 2023-11-07:
This is meant to be posted under https://github.com/wills106/homeassistant-solax-modbus/discussions/520
But that doesn't matter too much.
Main thing I need to know is what exactly is this Inverter as that's a true unique Serial Number so far.

Joerivanrokeghem, 2023-11-07:
Hi wills106,

the model of this inverter is a X1-Hybride-5.0K-D

wills106, 2023-11-07:
Just a normal Gen4? Or an updated model?

Joerivanrokeghem, 2023-11-07:
Yes, a normal Gen4

Met vriendelijke groeten,
Van Rokeghem Joeri

wills106, 2023-11-07:
Ok, I just wanted to double check it want a new Generation.
I'll try add it in tomorrow.

Joerivanrokeghem, 2023-11-09:
thanks in advance!

wills106, 2023-11-10:
Just released [2023.11.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.11.2)

wills106, 2023-11-29:
Can I close this down?

Joerivanrokeghem, 2023-11-29:
Yes it works, and can be closed. Thanks for the great work!

Met vriendelijke groeten,
Van Rokeghem Joeri

closed 2023-11-29 by wills106