wills106/homeassistant-solax-modbus #602: Feature Request - Bitwise Control of Solis Storage Control Switch

fboundy, 2023-11-08:
### Describe the bug

The Solis Storage Control Switch is current set using `select.solis_energy_storage_control_switch' and the options are limited to:

  - Selfuse - No Grid Charging (1)
  - Timed Charge/Discharge - No Grid Charging (2)
  - Selfuse (3)
  - Timed Charge/Discharge
  - Off-Grid Mode
  - Battery Awaken
  - Battery Awaken + Timed Charge/Discharge
  - Backup/Reserve

A more flexible approach would be to allow the individual bits to be set as individual `boolean` entities and then to calculate the final code from them:

[image]

### Integration Version

2023.11.1

### Homeassistant core version

2023.11.1

### Inverter brand

Solis

### Plugin used

solis.py

### First 6 digits of Serial (Growatt FIrmware Version)

N/A

### Connection Method

N/A

### I have done my own research

- [X] Yes, I did

wills106, 2023-11-10:
I have this as an Improvement idea https://github.com/wills106/homeassistant-solax-modbus/discussions/200 but it's something I have not got round to yet.

wills106, 2023-12-03:
Closing down as it's captured as an Improvement idea.

closed 2023-12-03 by wills106