wills106/homeassistant-solax-modbus #605: [Bug]: solax_export_control_user_limit incorrect

asuweb, 2023-11-08:
### Describe the bug

In the integration, I have set `solax_export_control_user_limit` to 9200W.

This is being sent to the inverter as 920W - it seems it is missing the trailing `0`

As a workaround I've disabled this entity until fixed

### Integration Version

2023.11.1

### Homeassistant core version

2023.10.1

### Inverter brand

Solax X1 Hybrid G4

### Plugin used

Solax Modbus

### First 6 digits of Serial (Growatt FIrmware Version)

H4502A

### Connection Method

Pocket WIFI v3 - V3.010.**

### I have done my own research

- [X] Yes, I did

cewbdex, 2023-11-08:
Works correctly on G4 Hybrid X3, strange. Please use solax form for solax issues... What is your firmware version?

wills106, 2023-11-10:
I'm guessing you are facing the same issue as https://github.com/wills106/homeassistant-solax-modbus/issues/418#issuecomment-1574806104

Might need to adopt the change mentioned for 'H4502T' Inverters on your 'H4502A' Inverter as well.

asuweb, 2023-11-14:
I can confirm it's the same issue. Thanks for your help. Setting `solax_config_export_control_limit_readscale` to `1` solved the issue and it now works as designed. 

Many thanks.

closed 2023-11-14 by asuweb