wills106/homeassistant-solax-modbus #609: Inaktive Funktionen

wartungsmonteur, 2023-11-11:
### Describe the bug

Einige Funktionen für die Peakshaving sind nicht nutzbar.

### Integration Version

2023.11.2

### Homeassistant core version

2023.11.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34C10

### Inverter firmware versions

ARM 2.03 DSP 2.07

### Connection Method

Waveshare 

### Dongle firmware

34 versteckte Entitäten für Gernerator und PeakShaving Modus

[image]

### Detailed Error Log

Fehlende Menüpunkte siehe Screenshot

wills106, 2023-11-11:
They are the read only sensors for PeakShaving and Generator control.

They are disabled by default as the number and select entities will provide the actual control and feedback. (There is no control at the moment)
You can manually enable them if you want to view if they are returning the correct set values.

I need to add in the number and entities though so you can control them, I have been too busy over the past couple of weeks to finish implementing them correctly.

wills106, 2023-11-11:
Just changed the status to enhancement, as it's something than needs finishing off.

I probably should have kept them in a side branch until I finished implementing them fully.

wartungsmonteur, 2023-11-11:
Vielen Dank für die tolle Arbeit

wills106, 2023-11-11:
Do you use Generator control? Or just the PeakShaving function?

wills106, 2023-11-11:
I have just checked the plugin_solax file...

I have added in the PeakShaving control, but you need to reconfigure the Integration and enable "Dry Contact Box (Gen4)"

I'll move PeakShaving back to the normal Gen4 configuration in the next release.

wartungsmonteur, 2023-11-11:
Ich nutze die PeakShaving Funktion. Es reicht eigentlich zu wenn man die Funktion PeakShaving Discharge Limit 1 und Discharge Limit 2 nutzen könnte.

wills106, 2023-11-11:
If you can try with "Dry Contact Box (Gen4)" and let me know if you have control.

I'll update the Integration hopefully tomorrow, so all the PeakShaving control is there by default.

wartungsmonteur, 2023-11-11:
Ich teste das, echt cool was du da machst.

wartungsmonteur, 2023-11-11:
Eben getestet, alles per Hand aktiviert und nutzbar, super.

wills106, 2023-11-11:
> Eben getestet, alles per Hand aktiviert und nutzbar, super.

Thanks for confirming, I'll correct the Integration so you don't need to enable DCB for the PeakShaving control.

Hopefully get a new release out tomorrow for Monday.

wills106, 2023-11-12:
Just released [2023.11.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.11.3) you shouldn't need to enable "Dry Contact Box (Gen4)" to be able to control the PeakShaving functions.

wartungsmonteur, 2023-11-12:
Hallo

Es fehlen zwei Schieberegler bei der PeakShaving Funktion.im Solax Plugin.

PeakShaving Discaharge Limit 1
PeakShaving Discaharge Limit 2

Es werden nur die Werte angezeigt. Man kann diese nicht im Home-Assistenten ändern.

wills106, 2023-11-12:
I have had a quick look I can't find an obvious write register for Discharge.

I'll have another more detailed look later.

wartungsmonteur, 2023-11-12:
https://photos.app.goo.gl/fFJVXZLXQzu5zuvMA

Hier die Option im Wechselrichter als Screenshot.

wills106, 2023-11-13:
> PeakShaving Discaharge Limit 1
> PeakShaving Discaharge Limit 2

I think I have found these.

Do the sensors return the correct value, or is the scaling out by x10

So it's 600 instead of 6000

wartungsmonteur, 2023-11-13:
Die Eingabe erfolgt in 1er Schritten, also 500 oder 501....

Der Bereich geht von 0 bis maximal 60000

wills106, 2023-11-13:
Just released [2023.11.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.11.4) check the release notes.

wills106, 2023-12-03:
Closing down as I believe the controls are all functional now.

closed 2023-12-03 by wills106