wills106/homeassistant-solax-modbus #610: [Bug]: Losing entities after restart HA

lieglw, 2023-11-12:
### Describe the bug

Hello,
first of all thank you very much for this integration.
I have 2 inverters, the Solax X3 Hybrid 15 (H34A) and the X3 Mic 4 (MC204).
I have connected them via Pocket Wifi to my system. IPs are fixed. I can add them easily to my HA.
The data is consistently available. I had a VM on a MAC and now Home Assistantt Yellow system.
But when I am shutting down the device (hard reset or soft) the data of the Hybrid is not available anymore. If I just remove the inverter of the integration and add it again only about 90 entities are showing up (e.g PV meassurement is missing).
I have to remove the inverters of the integration and then delete it. Re download it and then add the new hubs with the same name. Everything is working until the next shut down.
Have tried different COM addresses the same and different ones and tried different Boudrates.
nothing helped, always losing minimum the hybrid one.
Now have added all again have 251 entities for the hybrid normally I have 243.
The EV charger is also connected the Hybrid via com.

### Integration Version

2023.11.3

### Homeassistant core version

2023.11.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34A15

### Inverter firmware versions

ARM 2.03 DSP 2.07

### Connection Method

Solax Pocket WIFI

### Dongle firmware

29 09 27

cewbdex, 2023-11-12:
The fw version you provided is inverter's, not of dongle. and the inverter ones are invalid, read the wiki linked in readme for more information.

wills106, 2023-11-12:
Are you restarting the Integration / Home Assistant while the Hybrid is sleeping / idle mode?
If so that will prevent the Integration establishing what Inverter it is, you need to make sure the Inverter is in Normal mode before restarting Home Assistant.

Firmware version details are mentioned [here](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq). But the readme needs further clarity
They are disabled by default, you can read how to enable them [here](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#entity-i-need-is-not-present)

lieglw, 2023-11-12:
V3.006.04 Wifi Dongle 
29 09 27 Inverter

Restarts happened in different situation as I'm upsetting the HA system in the moment.
It's more like having all the data and need to restart because of the an update (eg). Then the data from the entity is not 'unkown' but 'unavailable'. The 'unknown' will update for sure.

wills106, 2023-11-12:
Can you let me know what is outputted from your error log once you have restarted Home Assistant
```
If there are no log's showing for SolaX try the following depending on your language settings, you can find the full logs under:
  Settings → System → Logs > at bottom of page press “LOAD FULL LOGS”
Now the full logs are loaded. If you scroll down, you will see them. Once the full logs are shown, you can either use the search function in your browser to search for “solax” related entries or use the search entry field on top of the page.
Search for
  solax
and report us the logs. Make sure to replace sensitive information by xxxx (if any)
```
If you still don't have anything you might need to add:
```
logger:
  default: info
```
Into your configuration.yaml

The ARM 2.03 DSP 2.07 on the Inverter LCD is static, it doesn't update even if you have newer software on the Inverter.
Or it might be a static bootloader version? As the Gen4 returns the same values as the Gen3.

Gen3 is on v3.xx when read from the Integration
Gen4 is on v1.xx when read from the Integration

cewbdex, 2023-11-12:
Read the <https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/>, try upgrading to later dongle fw.
Do both inverters not work after restart or just one?  I am running X3 Hybrid G4 with Pocketwifi 3.0 and restarts are no issues, running for two months with no issues. Also you did not specify the version of PocketWiFi, so I can't be certain.

lieglw, 2023-11-12:
Thank you.
Made a restart: Both worked direct before.
After restart Mic loaded the available entities as we have no sun anymore. For the Hybrid it shows that the values are unavailable.
[image]
[image]
[image]
That’s the log after restart:

2023-11-12 16:35:25.223 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-12 16:35:25.225 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration multimatic which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-12 16:35:25.228 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-12 16:35:25.230 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration husqvarna_automower which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-12 16:35:40.297 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX Mic: reading serial number from address 0x0 failed; other address may succeed
2023-11-12 16:35:51.378 WARNING (MainThread) [zigpy.application] Zigbee channel 25 utilization is 95.12%!
2023-11-12 16:35:51.378 WARNING (MainThread) [zigpy.application] If you are having problems joining new devices, are missing sensor updates, or have issues keeping devices joined, ensure your coordinator is away from interference sources such as USB 3.0 devices, SSDs, WiFi routers, etc.

lieglw, 2023-11-12:
Sorry this was directly before restart
[image]

wills106, 2023-11-12:
Can you add:
```
logger:
  default: info
```

To configuration.yaml and go through the process again. You only have the basic logging enabled at the moment.

When you configured the Integration did you give the Inverters unique names.
Like SolaX MIC & SolaX Hybrid, as long as they aren't both named SolaX.
SolaX & SolaX2 is ok, or SolaX1 & SolaX2 etc

lieglw, 2023-11-12:
Yes named them separately:
Solax Hybrid 
Solax Mic
Will make the change and process it. Can send it at night

wills106, 2023-11-12:
The Mic will fail to connect if done at night as it will have powered down due to not enough PV Voltage to keep it alive.

lieglw, 2023-11-12:
Never had this problem, it just shows 0

lieglw, 2023-11-12:
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:66 
Integration: SolaX Inverter Modbus (documentation, issues) 
First occurred: 17:26:56 (1 occurrences) 
Last logged: 17:26:56

Funny thing, the mic works.

SolaX Mic: reading serial number from address 0x0 failed; other address may succeed
[home-assistant_2023-11-12T16-29-13.354Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/13328573/home-assistant_2023-11-12T16-29-13.354Z.log)

lieglw, 2023-11-12:
Tried now the Hybrid only and reboot is no problem

lieglw, 2023-11-12:
Added the Mic and restarted, entities from Hybrid are unavailable again.

wills106, 2023-11-12:
> SolaX Mic: reading serial number from address 0x0 failed; other address may succeed

That's actually an ok error, the MIC's serial is on 0x400.
So it fails on 0x0 but finds it on 0x400.

Looking at the log it reads the Serial for both your Hybrid and Mic, but it's only building a register set for the Mic.

lieglw, 2023-11-12:
When added Mic first and then the Hybrid only 60 entities from the Hybrid show up. 
[image]

Then again the other way round:
[image]

wills106, 2023-11-12:
> Have tried different COM addresses the same and different ones and tried different Boudrates.

Do both Inverters have a PocketWiFi?
Or is one using a PocketWiFi and the other a UBS-RS485 adaptor?

If both are using a PocketWiFi there shouldn't even be an option to set a Baudrate.

lieglw, 2023-11-12:
Both have a Pocket Wifi

lieglw, 2023-11-12:
I have the EV Charger connected to the hybrid via COM

lieglw, 2023-11-13:
Today both inverters received an update by Solax. Tested the restarting issue. New firmware has no impact on the issue.

lcavalli, 2023-11-13:
@lieglw, are you sure that your DHCP server didn't assign a different IP address? In this case, simply reconfigure the integration in HA.

lieglw, 2023-11-14:
> @lieglw, are you sure that your DHCP server didn't assign a different IP address? In this case, simply reconfigure the integration in HA.

No IP is fixed. Because deleting everything (not HA) and installing again works with theme same IP

lieglw, 2023-11-14:
> Read the https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/, try upgrading to later dongle fw.
> Do both inverters not work after restart or just one? I am running X3 Hybrid G4 [...]

Both inverters were updated and also the Pocket WiFi is on V3.006.04.
Problem is only when having the 2nd inverter (MIC) added.

lieglw, 2023-11-14:
> > SolaX Mic: reading serial number from address 0x0 failed; other address may succeed
> 
> That's actually an ok error, the MIC's serial is on 0x300. So it fails on 0x0 but finds it on 0x300.
> 
> Looking at the log it reads the Serial for both your Hybrid and Mic, but it's only building a register set for the Mic.
> 
> Edit: Corrected MIC serial to be 0x300 and not 0x400

@wills106 : should I change anything in the seetings when adding the inverters?

lieglw, 2023-11-26:
Hi, hi today we had a grid fall out since than I have this error. Perhaps this also helps for this issue 
Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:6047
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 11:54:40 (1 occurrences)
Last logged: 11:54:40

SolaX Hybrid : cannot find serial number, even not for MIC

lieglw, 2023-11-29:
Hi, any idea on this?

wills106, 2023-11-29:
You have lost the Modbus connection to the PocketWiFi
Have you tried what's listed under https://github.com/wills106/homeassistant-solax-modbus#pocket-wifi-30-with-firmware-v300403-and-above-is-only-officially-supported-solax-only-mentions-gen4-hybrid-other-inverters-may-work
Also you might need to ensure your Inverters are actually producing enough PV for them to respond to Modbus. I know on the RS485 connection that's the case. I'm unsure about PocketWiFi as never used one.

lieglw, 2023-11-29:
Hi, as long as the system is running it works. But as soon as the system restarts  I am losing the entities if I have both inverters connected as soon as I use only one inverter I can restart without issue. 
PV is no problem as long as it is not in idle mode. So if there is still battery capacity left connection works.

only have lost Modbus connection one time. All the other times I needed to remove devices from integration and remove integration install it again add the devices and than it works.
If only 4 entities appear after this modbus connection is lost and I have to unplug the dongle.
Main issue is having 2 inverters (of course with different names) and restarting.
With the new HA core 2023.11.3 no connection is possible at the moment only getting 4 entities.

Working HA core 2023.11.2
Thanks Wolfgang

wills106, 2023-11-29:
> With the new HA core 2023.11.3 no connection is possible at the moment only getting 4 entities.

I am wondering if you are facing a similar issue to #619 Can you check if you have pyModbus installed when you are on HA 2023.11.3 Are you using HA Core in Docker? Or a variant of HAOS?

I haven't experienced #619 on my Install, I am using HA Core in Docker. But I am not using any PocketWiFi devices.

lieglw, 2023-11-29:
Hey I am not that familiar, but I have read a bit about it. It seems that I am not using a container.
It runs on a HA yellow.
[image]
It also seems that your most stable version is 2023.11.3
You need to guide me how to check for pyModbus

wills106, 2023-11-29:
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#how-to-check-a-version-of-a-python-module-eg-pymodbus

I you need to do anything specific on the Yellow I don't know? Might be best asking in the HA forum.

lieglw, 2023-11-29:
So checked with pip list and found no Modul named pyModbus.
But this time the upgrade to 2023.11.3 worked, but with integration version 2023.11.3 and I removed the mic before I upgraded HA.

lieglw, 2023-12-01:
@wills106 Could it be that the error only occurs if connecting 2 inverters via WiFi.
As long as I just have one of the inverters connected (does not matter which one) can restart without any problems.

wills106, 2023-12-01:
I don't know. I only have a single Inverter with built in Ethernet.

I know the Integration works ok with two Gen2 Hybrid Inverters which also have built in Ethernet.

I can't perform any further testing on the issue unfortunately.

cewbdex, 2023-12-01:
I have experienced an odd behaviour, but no long term problems with SolaX Wi-Fi adapter. When I enable a disabled entity, after a while, all entities get unknown, after 15 sec (my polling rate) they all appear back. May be related.

lieglw, 2023-12-02:
> I don't know. I only have a single Inverter with built in Ethernet.
> 
> 
> 
> I know the Integration works ok with two Gen2 Hybrid Inverters which also have built in Ethernet.
> 
> 
> 
> I can't perform any further testing on the issue unfortunately.

Could it be that by using 2 WiFi dongle that the second SN overwrites the first in the register?

emrares, 2023-12-03:
Having this issue on my X3-Hybrid-G4 and HA 2023.11.3

After restart the entities become unknown.

Also I noticed that after a while the entites do not update.

For example the inverter temperature after a while is showing the value zero.

treimy, 2024-02-29:
I have the issue that after restarting de pi, or container or Home Assistant, all Entities are showing no values. Before it was fine. This is happening at day, wenn the X3 Mic is in normal mode and it also happens at night. Firtly i tried the WiFi 3.0 Dongle and now with the Wifi+LAN Dongle. The behaviour is the same.

The trick for me is to update the configuration of the device, i.e. change the update interval. After that, all values are shown and updating again.

Thats very annoying to do that every restart...

lieglw, 2024-04-22:
@wills106 : Updated system and integration today. HA OS 12.2, Core 2024.4.3 and integration 2024.04.3.
Summary: Using Solax Hybrid 15kW and a Mic 4. Since Nov 2023 no combination of HA and integration result in a restart without losing the entities of min one of the inverter. This system now provides stable restarts. Nothing was changed about connection of the inverters. Did 2 restarts now and everything works great.
Only one thing leads to a problem, activating the entity "Mic PV Total Power" leads to losing all entities for the MIC. Deactivating this entity and restarting HA --> all ok again.
Perhaps this helps.

wills106, 2024-09-20:
Is this still an issue on later Integrations such as 2024.09.x? Or can this be closed down?

lieglw, 2024-09-22:
With the 2024.09.1 currently using its more stable, but cannot test as the entities currently do not really update. They are long unknown and there is an error in the data.
Using 2 instances one on a ha yellow with 2024.05.7 just with the Mic added shows e.g sum of pv power very accurate. And shows after restart very fast values.
On the second instance have both inverters restart issue seems to be gone but values are wrong or just show very long unknown.

lieglw, 2024-11-14:
@wills106 Since I have both inverters on 2 seperate HA. This no issue anymore.

wills106, 2024-11-16:
I'll close down then

closed 2024-11-16 by wills106