wills106/homeassistant-solax-modbus #619: Integration reports entities as unknown after upgrading HA to version 11.0 or 11.1 or 11.2

funks81, 2023-11-13:
I've tried upgrading HA core to version 11.0 / 11.1 / 11.2 and after upgrading the entities are "unknown".  On downgrading back to HA core 10.5 everything starts working again.  I get the error reading serial number when upgrading HA core - but find this strange as everything seems to run fine whilst running HA core 10.5.

Integration Version: 2023.11.3
HA Version: 2023.11.0 / 11.1 / 11.2
Inverter Brand: Solis
Plugin Used: Solax Modbus
First 6 digits of Serial: 180102
Connection Method: Modbus via TCP

I have the following errors in my log when running any of the HA core 11.0/11.1/11.2 versions:

2023-11-13 12:32:21.210 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-13 12:32:31.203 WARNING (MainThread) [custom_components.solax_modbus.plugin_solis] SolisDirect: reading serial number from address 0x80ec failed; other address may succeed
2023-11-13 12:32:31.204 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] SolisDirect: cannot find serial number, even not for other Inverter
2023-11-13 12:32:31.204 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis] unrecognized SolisDirect inverter type - serial number : unknown
2023-11-13 12:33:38.749 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform solax_modbus
2023-11-13 12:33:38.755 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solax_modbus platform for sensor

When running HA core 10.5 the only thing showing in my log is:

2023-11-13 11:15:26.020 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

wills106, 2023-11-13:
I have just gone from HA 2023.10.x to 2023.11.2 perfectly fine.
Once you have moved to 2023.11.2 did you try restarting HA again?

DKen71, 2023-11-13:
I am having the same issue. Restarted HA multiple times. Checked I had the latest Solax modbus installed and reloaded. Checked the logger. I don't have a back-up old enough to roll back to HA10.x.

wills106, 2023-11-13:
Do you get anymore information in the error log if you add:

```
logger:
  default: info
```

to the configuration.yaml file?

When you are on HA 2023.11.2 what version is pyModbus showing? They moved to 3.5.4 from 3.5.2, but has moved over to 3.5.4 correctly.

You can find out by following [Point 4 of FAQ](https://github.com/wills106/homeassistant-solax-modbus#faq)

Are either of you using any other Integrations using Modbus?

funks81, 2023-11-13:
Yep - did try restarting HA again after the upgrade - will try the logger when I get home - thanks. Sent from my iPhone

DKen71, 2023-11-13:
I added the additional YAML to the configuration.YAML. Restarted. No additional data provided.

I only use Modbus for the inverter - nothing else.

wills106, 2023-11-13:
@DKen71 can you have a look at what version pyModbus is installed. Wonder if it's got corrupt or completely missing?

DKen71, 2023-11-13:
@wills106 It is not listed at all. I have just upgraded to 2023.11.2 to see if that solved it, but still missing. Not sure where it went between upgrades.

wills106, 2023-11-13:
Have you checked pyModbus is still there after reinstalling the integration and restarting HA?

Just wondering if there is a HA bug causing the python module to get deleted?

funks81, 2023-11-13:
So - just upgraded again to latest version HA core - with logger turned on in config

When I first restarted - I briefly got data for the entities - then went to unknown again - have restarted HA for a second time - and currently seem to be getting data - only been up for about 5 minutes - but seems to be working.

Have just manually restarted the integration and now getting "Unavailable" beside entities -but interestingly no errors shown in the logs.

Have tried restarting the EW11 - remains unchanged

After restarting HA for a third time - entities showing as Unavailable - have put the logs for this restart at the bottom of this page

Not sure where I'm looking for pyModbus - couldn't see in HACS or HA?

Getting the following in the logs with logger switched on:

2023-11-13 15:16:25.387 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-11-13 15:16:29.364 INFO (MainThread) [homeassistant.bootstrap] Setting up stage 2: {'homeassistant', 'script', 'mobile_app', 'media_source', 'samsungtv', 'scene', 'select', 'hardware', 'solax_modbus', 'map', 'history', 'image_upload', 'solcast_solar', 'auth', 'logbook', 'octopus_energy', 'automation', 'tag', 'energy', 'file_upload', 'config', 'application_credentials', 'repairs', 'onboarding', 'sensor', 'tuya', 'counter', 'zone', 'wake_word', 'my', 'diagnostics', 'localtuya', 'stt', 'conversation', 'tts', 'input_number', 'input_boolean', 'tesla_custom', 'meross_lan', 'sun', 'hacs', 'default_config', 'blueprint', 'person', 'schedule', 'uptime', 'homeassistant_alerts', 'input_button', 'persistent_notification', 'ffmpeg', 'mqtt', 'alexa', 'speedtestdotnet', 'input_datetime', 'timer', 'esphome', 'fully_kiosk', 'input_text', 'integration', 'threshold', 'heatmiserneo', 'analytics', 'shelly', 'stream', 'forecast_solar', 'device_automation', 'assist_pipeline', 'utility_meter', 'trace', 'lovelace', 'search', 'system_health', 'input_select'}
2023-11-13 15:16:29.792 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2023-11-13 15:16:29.792 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2023-11-13 15:16:29.920 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2023 11
2023-11-13 15:16:29.920 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2023-11-13 15:16:29.921 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solis
2023-11-13 15:16:29.928 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] SolisDirect: trying to determine inverter type
2023-11-13 15:16:30.322 INFO (MainThread) [custom_components.solax_modbus.plugin_solis] Read SolisDirect 0x80ec serial number: 60310502273002, swapped: False
2023-11-13 15:16:32.381 INFO (MainThread) [homeassistant.components.select] Setting up select.solax_modbus
2023-11-13 15:16:32.393 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa7ff power_switch to block with start 0xa7ff
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa803 battery_minimum_soc to block with start 0xa7ff
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa810 backup_mode_soc to block with start 0xa7ff
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa841
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa841 backflow_power_switch to block with start 0xa841
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa842 backflow_power to block with start 0xa841
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86c battery_chargedischarge_current to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86d battery_charge_current to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa86e battery_discharge_current to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa885 timed_charge_current to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa886 timed_discharge_current to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa887 timed_charge_start_h to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa888 timed_charge_start_m to block with start 0xa86c
2023-11-13 15:16:32.394 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0xa889 timed_charge_end_h to block with start 0xa86c
[... cut ...]
2023-11-13 15:29:48.408 INFO (MainThread) [homeassistant.components.sensor] Setting up sensor.solax_modbus
2023-11-13 15:29:48.409 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {}
2023-11-13 15:29:48.410 INFO (MainThread) [homeassistant.components.button] Setting up button.solax_modbus
2023-11-13 15:29:48.411 INFO (MainThread) [homeassistant.components.number] Setting up number.solax_modbus
2023-11-13 15:29:48.411 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None

DKen71, 2023-11-13:
Something very strange happening with my set-up.

I installed pymodbus, then confirmed it is installed. If I restart HA, then it is still there. If I reboot HAOS, then it disappears. 

I've done this twice now and I get the same results. 

Looks like I may have to rebuild this.

DKen71, 2023-11-13:
I've changed requirements. On restart of HA, it retains pymodbus.3.4.5. Restarting HAOS deletes it again.

funks81, 2023-11-13:
I'm running HAOS on a VM in Proxmox - pip doesn't seem to be a supported command from the console for HA

DKen71, 2023-11-14:
> does it survive restarts of HA or does it still disappear?

It doesn't survive the restart of the OS.

However, I have everything back and working now. I rolled back the Core to 2023.10.5 and reinstalled SolaxModbus via HACS to a 2023.10.2. That still was not working. I then upgraded each component in sequence. Core 2023.11.1 and SM 2023.11.1 gave me the functionality back. 

I followed on that process and now I am up to date on both components and it is still working.

Having said that, pymodbus 3.4.5 is still not showing when I use `pip show pymodbus` or `pip list`.

Not sure what I have done, but there must have been something that broke during an upgrade. I'll report back if anything else crops up. 

Thanks for the patience and assistance.

closed 2024-01-10 by wills106