wills106/homeassistant-solax-modbus #620: [FR]: Add Solis Force Charge SOC value entity, like Battery Minimum SOC

0rangutan, 2023-11-13:
### Describe the bug

The entity number.solis_waveshare_battery_minimum_soc is presented in the integration as a slider that can be adjusted to change the setting for the inverter's minimum battery SOC value.  This works really well, thanks.
Would it be possible to add the same for the battery's Force Charge SOC value?
This is present in the Solis Cloud integration (eg. sensor.solis_force_charge_soc) but is not available in the 145 entities exposed for my inverter (including the disabled ones).

### Integration Version

2023.11.3

### Homeassistant core version

2023.11.2

### Inverter brand

Solis

### Plugin used

plugin_growatt.py

### First 6 digits of Serial (Growatt Firmware Version)

603105

### Connection Method

RS485-Eth (B) Waveshare dongle

wills106, 2024-01-10:
I know where about the Force Charge SOC is based on the change log.
But I don't have access to the actual section of the Modbus registers indicating exactly which one does what function and I don't know if they need to be written in sequence like "Update Charge/Discharge Times"

I can try and implement the Force Charge SOC percentage by it's self. But there is a risk you might alter something else by mistake.

0rangutan, 2024-01-10:
Force Charge SOC is implemented successfully in this integration if it helps?
https://github.com/Pho3niX90/solis_modbus

wills106, 2024-01-10:
Ok so it is just a single register 43018 like I was hoping it would be.

wills106, 2024-01-23:
This was added in [2024.01.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.6) did you manage to try it? If so can you close this issue down.

wills106, 2024-02-12:
Closing down as no further feedback.

closed 2024-02-12 by wills106