wills106/homeassistant-solax-modbus #621: [Bug]: holding register already used: 0x147 generator_min_power

michaelschwarz, 2023-11-13:
### Describe the bug

I am receiving following error in logs since last update.

> Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht
> Logger: custom_components.solax_modbus.sensor
Source: custom_components/solax_modbus/sensor.py:115
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 16:35:17 (2 occurrences)
Last logged: 16:39:38
> holding register already used: 0x147 generator_min_power

The inverter Q.VOLT HYB-G3-3P is connected using TCP/IP interface.

### Integration Version

2023.11.2

### Homeassistant core version

2023.11.2

### Inverter brand

Qcells

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34B12

### Inverter firmware versions

-

### Connection Method

SolaX Pocket WiFi 3.0

### Dongle firmware

-

wills106, 2023-11-13:
Just released [2023.11.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.11.4)

michaelschwarz, 2023-11-13:
Looks like this issue is solved, thanks for this quick fix!

wills106, 2023-11-13:
Do you actually use any of the Generator control or Dry Contact Box stuff?

michaelschwarz, 2023-11-13:
I just received the new inverter and battery, and reviewing what options are there. Except of setting the self use battery capacity I didn't use any method/configuration yet.

Is there something that I can test?

wills106, 2023-11-13:
I was just going to ask if you actually use Generator control or Dry Contact Box if you could write some basic documentation. As I have no way of testing.

I'll let you get used to your Inverter as it's a new Install 😉
So don't worry about it.

If you find any other issues let me know.

I'll close the issue down as you have no more errors. Thanks for letting me know about that error.

closed 2023-11-13 by wills106