wills106/homeassistant-solax-modbus #622: [Bug]: PV DC Power is not exposed for individual PV strings on Solis inverter

0rangutan, 2023-11-13:
### Describe the bug

The sensor for pv_total_power is presented in the integration for my Solis inverter (3105, S5-EH1P6K-L) but **not** the separate DC PV Power entities for strings 1 and 2.  The PV Voltage and Current entities **are** provided for strings 1 and 2.  The two strings are on East and West facing sides so being able to measure them independently is really useful.
These are exposed in the Solis Cloud integration as sensor.solis_dc_power_pv1 and pv2 and this is the only thing I am still using from that integration - would love to drop it.
Are these values available on other inverters?  Or is the Watts power per string typically just a calculated value?
If the latter, would you consider adding per-string calculation of the Watts Power as a feature request?

### Integration Version

2023.11.3

### Homeassistant core version

2023.11.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### First 6 digits of Serial (Growatt Firmware Version)

603105

### Connection Method

RS485-Eth (B) Waveshare dongle

wills106, 2024-01-10:
It seems on the Solis there aren't separate registers for PV1 & PV2 Power etc, which is a bit strange as other Brands report this.

0rangutan, 2024-01-10:
Yes, I believe that's the case.
Would need to be calculated values then - not sure how that would work if you have actual values for other inverters.

wills106, 2024-01-10:
Just released [2024.01.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.4) with computed values.

0rangutan, 2024-01-10:
Fantastic, thanks

wills106, 2024-01-10:
@0rangutan have you been able to test this yet or do you get #693

0rangutan, 2024-01-10:
Tested on 2024.01.5 and not experiencing that bug

wills106, 2024-01-10:
It got fixed in 2024.01.5

I'll close this down now.

closed 2024-01-10 by wills106