wills106/homeassistant-solax-modbus #627: [Bug]: Added "Safety code"

matomatusov, 2023-11-14:
### Describe the bug

 Hi,
"Safety code" does not display the correct description.
Solax application - Czech_2022_01
Entity SolaX Safety code  Czech_PPDS_2020

### Integration Version

2023.11.4

### Homeassistant core version

core-2023.11.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34A10

### Inverter firmware versions

ARM1.35 DSP1.34

### Connection Method

Waveshare adapter

### Dongle firmware

Solax Packet Wifi 3.010.16

wills106, 2023-11-14:
There is a conversation https://github.com/wills106/homeassistant-solax-modbus/pull/599#issuecomment-1809205018 about a similar mismatch.
I don't know if when you get a newer Firmware version it will report something different in the cloud or Inverter LCD for the same result from the register.

I don't have access to a Gen4 to cycle through every Safety Code and I don't know if a Firmware update would change it. I can only go off what's in the Docs. If I had access to every revision of the docs I could see if it does actually change or not.

matomatusov, 2023-11-14:
Safety Code does not affect the firmware. I've had it since the beginning of the installation. I have already updated the firmware several times.

Thank you for the information.

closed 2023-11-14 by matomatusov