wills106/homeassistant-solax-modbus #629: [Bug]: "solax_inverter_current" value is the same as of "solax_inverter_current_l1"

mbeetle, 2023-11-16:
### Describe the bug

The Sensor solax_inverter_current has the value of the Sensor qcells_inverter_current_l1 instead of the sum from
solax_inverter_current_l1 + solax_inverter_current_l2 + solax_inverter_current_l3

### Integration Version

2023.11.4

### Homeassistant core version

2023.11.2

### Inverter brand

Q.VOLT_HYB-G3-3P

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34C12

### Inverter firmware versions

ARM 1.33 DSP 1.34

### Connection Method

Waveshare Adapter

### Dongle firmware

Modbus

wills106, 2024-02-20:
I know this got raised first. But is it the same as #667

mbeetle, 2024-02-20:
Hi Will

Yes it is the same

closed 2024-02-20 by wills106

mbeetle, 2024-02-21:
> Ok, I'm going to close this down then as #667
> <https://github.com/wills106/homeassistant-solax-modbus/issues/667> has
> had more recent activity.
>
> I'm trying to organise and consolidate issues so we (As in the community)
> can better focus on them.

Hi Will
I am sorry, i did not read correctly. It is not the same. My bug was
Inverter current, not Inverter Power.