wills106/homeassistant-solax-modbus #630: [Question]: Solax_ev_charger

matomatusov, 2023-11-18:
### Describe the bug

Hi,
is this integration working? I can't load the entities.

Thanks

[image]

### Integration Version

2023.11.4

### Homeassistant core version

core-2023.11.2

### Inverter brand

Solax Power

### Plugin used

plugin_growatt.py

### First 6 digits of Serial (Growatt Firmware Version)

H34A10

### Connection Method

Waveshare adapter

PatrikTrestik, 2023-11-23:
It depends how you connect it. 
EV Charger has no Modbus TCP option so you have to connect it by wire and Modbus RTU.
But than you cannot connect it ma Modbus to Inverter for smart functions.

We are currently in the middle of testing HTTP API access, but it is in early stage.

matomatusov, 2023-12-01:
I have it connected to the inverter on the COM port (Modbus RTU).

matomatusov, 2023-12-13:
Hi,
I am closing this bug.
It is solved through SolaX http.
https://github.com/PatrikTrestik/homeassistant-solax-modbus

Thanks.

closed 2023-12-13 by matomatusov

PatrikTrestik, 2023-12-13:
> Hi, I am closing this bug. It is solved through SolaX http. https://github.com/PatrikTrestik/homeassistant-solax-modbus
> 
> Thanks.

Correct link to http api integration: https://github.com/PatrikTrestik/homeassistant-solax-http