wills106/homeassistant-solax-modbus #635: [Bug]: as log file ask for :-)

modwadd, 2023-11-20:
### Describe the bug

report as requested by youre log file:
Logger: homeassistant.util.async_
Source: util/async_.py:140
First occurred: 14:57:38 (1426 occurrences)
Last logged: 15:11:56

Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

### Integration Version

latest

### Homeassistant core version

latest

### Inverter brand

Solax X3 G4

### Plugin used

plugin_solax.py

### First 6 digits of Serial

n.a

### Inverter firmware versions

n.a

### Connection Method

sereial to RS485 at GPIO COM of Pi4 

### Dongle firmware

connected to modbus com port

### Additional context

if the serial and firmware versions needs to be known ich could serch them

cewbdex, 2023-11-20:
Are you using USB to RS-485 adaptor?

modwadd, 2023-11-20:
I using this:https://www.waveshare.com/rs485-can-hat.htmDirect connected to com Port rs485 of solax

cewbdex, 2023-11-20:
Probably a duplicate of #379 
this hat is not officially supported by this integration, I believe it has the same problem as issue mentioned above.

wills106, 2023-12-03:
Closing down as a duplicate of #379

closed 2023-12-03 by wills106