wills106/homeassistant-solax-modbus #637: [Bug]: writing solis backflow_power

jojoBzh, 2023-11-22:
### Describe the bug

Hello,
I'm encountering the following error on the Solis S5-EH1P3K-L inverter when attempting to modify the Backflow Power:
"struct.error: 'h' format requires -32768 <= number <= 32767"
Thank you for your valuable work and for sharing this information. It's exactly what I was looking for.

Best regards,

### Integration Version

2023.11.4

### Homeassistant core version

2023.11.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### First 6 digits of Serial (Growatt Firmware Version)

103101

### Connection Method

GC-1201K RS485 / Modbus

### Detailed Error Log

2023-11-22 15:40:53.601 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547616742464] 'h' format requires -32768 <= number <= 32767
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 106, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/solax_modbus/number.py", line 147, in async_set_native_value
    self._hub.write_register(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 366, in write_register
    if awake: return self._lowlevel_write_register(unit, address, payload)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 358, in _lowlevel_write_register
    builder.add_16bit_int(payload)
  File "/usr/local/lib/python3.11/site-packages/pymodbus/payload.py", line 197, in add_16bit_int
    self._payload.append(pack(fstring, value))
                         ^^^^^^^^^^^^^^^^^^^^
struct.error: 'h' format requires -32768 <= number <= 32767

cewbdex, 2023-11-22:
`modify`, how did you modify it? Which number etc.

jojoBzh, 2023-11-23:
Hello,

I've been manually adjusting the powerflow value using the "variation button" on the dashboard.
After conducting further tests:
- When manually set powerflow to 0W : the inverter operates correctly.
- When set to a small value (e.g., 100W) : the inverter displays the maximum power flow value of 9900W.
- When set to a larger value (e.g., 1000W) : it results in an out-of-range error message as described.
It appears to be a calibration issue with the button.

Regards,

jojoBzh, 2023-11-23:
If I set 0.1 via automation, the inverter displays 1000. Therefore, it seems that the set value is being multiplied by 10,000.
Thank you.

cewbdex, 2023-11-23:
If you are experienced user, you can try to play with this value: https://github.com/wills106/homeassistant-solax-modbus/blob/5730eb7e1228aae86a3886abcbf5ad5c18f3163a/custom_components/solax_modbus/plugin_solis.py#L608
This is the multiplier that gets applied to the number. I do not have the required documents to fix this, Will may have, but he seems busy now.

jojoBzh, 2023-11-23:
This OK with this value :
scale = 100,
Thanks a lot for your responsive help

cewbdex, 2023-11-23:
I believe this issue should be remain open, as this is something that should be investigated further by Solis maintainers (either some inverters use different scale factors or this was just a typo).

jojoBzh, 2023-12-05:
plugin_solis.py#L608 :
actual value : 
scale = 0.01,
to be corrected by :
 scale = 100,

Tested on my Solis inverter.
Thanks a lot

wills106, 2024-02-12:
I have just released [2024.02.3b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b4) with the change.

wills106, 2024-02-12:
Just released it [2024.02.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3) so closing down.

closed 2024-02-12 by wills106

jojoBzh, 2024-02-12:
Great, that's OK for me ;-)
Thanks a lot !
Best regards