wills106/homeassistant-solax-modbus #638: [Bug]: Please be so kind to add the new Solax X1 mini G4 to the serial number list

ThomasBlome, 2023-11-24:
### Describe the bug

Please be so kind the add the serial number the the Solax modbus integration

SOLAX X1 mini G4 1.5KW

Serial number 	XM4015Jxxxxxxx. (First 7 Digits provided), new model arriving on the German market

I did it manually in the script and assigned a to a x1 mini G2 and works so far, need to check the details values.

Important: it does not work in my installation with Modbus via RS485, instead I used the Network address for the Dongle Wifi 3, and it works like charming

Best regards

Thomas

### Integration Version

2023.11.1

### Homeassistant core version

2023.1.1

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### First 6 digits of Serial

 XM40

### Inverter firmware versions

ARM 1.37

### Connection Method

Solax Pocket WifI 3 dongle, see comment regarding RS485

### Dongle firmware

...

### Additional context

...

ThomasBlome, 2023-11-24:
in the meantime I checked the data extracted via the ModBus Wifi Dongle access, get quite some datasets (24 entities), however most of them are different values than the SolaxCloud data from the inverter, assume that teh integration needs to take on that.

wills106, 2023-11-27:
The Mini only has a limited set of Registers compared to the Hybrids, so 24 entities sounds right.

So you have an example of the differences? Your cloud Instance might only be updating every 5mins so that could be why you are seeing different values.

ThomasBlome, 2023-11-27:
Hi,

thank you very much for your fast response, I was not able to get any meaningful data over the modbus protocol of the mini X1 G4. The modbus itself is working because the solax communicated with an Chint DDSU666 energy meter pretty nicely.
Instead I was able to get the data via https://community.home-assistant.io/t/solax-x1-boost-air-mini-local-data/535197.
The G4 mini has a slightly different assignment of the addresses, that’s what I’m currently analyzing over this week, looks very promising.

Best regards

ThomasBlome, 2023-11-27:
Hi,

short update, got a message from the Solax Service Team in Europe, the WiFI 3 firmware was updated to the following version

Wi-Fi dongle
Pocket WiFi 3.0
V3.004.03

For the X1 mini G4 the data only could be accessed via modbus TCP/IP, RS485 is not supported because it used to communicate with the CHINT Power meter, per definition 2 Master and 1 Slave are not supported in the ModBus Specification

Will spend some time tomorrow to give it a try.

Best regards

Thomas

ThomasBlome, 2023-12-04:
The Solax X1 mini G4 is now recognized by the Modbus Integration, Baurate is 9600Baud. 1 Device and 26 entities are shown in the integration. However the assignment of the values is pretty strange, most of the values are not matching the real data. Only about 2 values (example:  inverter voltage) seems to be right. First thought is that the X1 mini G4  is not supporting the Modbus variables from the SOLAX Modbus whitepaper. Need to contact the Solax Service in Europe to find it out, keep the forum updated

cewbdex, 2023-12-04:
Where do you get the real data?

ThomasBlome, 2023-12-05:
attached the example with works for my Solax X1 mini G4 based on GitHub forum but modified for that type of inverter

you need to add your IP address for your inverter at: @your IP adress

you need to add serval times the serial number of your Wifi Dongle connected to your inverter: @ yourSerialnumber of wifi Dongle to be included here

The code needs to be included in the configuration.yaml file via the editor

There are still a few variables send by the inverter which are not assigned in the code (due to a lack of my time currently)

Code is build as a module, Solax1. so if you have more tan one just copy and paste and exchange Solax1 to Solax 2 etc.

Hope this helps you starting your project

#########################################################
#                                                       #
#         SOLAX.   Real Time Data via WifI 3 Dongle     #
#                                                       #
#                 SOLAX X1 mini G4                      #
#                                                       #
#                  Thomas J. Blome                     #
#                                                       #
#                  November 2023                        #
#                                                       #
#                    Version 2.1                        #
#########################################################
#
# ADR 00 Daten: AC Output Voltage   Format
# ADR 01 Daten: AC Current (INV).   Format
# ADR 02 Daten: AC Frequency.       Format
# ADR 03 Daten: AC Power.  (INV)    Format 
# ADR 04 Daten: DC Voltage (PV)     Format
# ADR 05 Daten:    Format
# ADR 06 Daten:    Format
# ADR 07 Daten:    Format
# ADR 08 Daten: DC Current (PV)     Format
# ADR 09 Daten:    Format
# ADR 10 Daten:    Format
# ADR 11 Daten:    Format
# ADR 12 Daten:    Format
# ADR 13 Daten: DC Power (PV).       Format
#
# ADR 19 Daten: Total DC Power.    (PV).       Format
# ADR 21 Daten: Total DC Power day (PV).       Format
# ADR 22 Daten: AC Export Power        .       Format
# ADR 23 Daten: Temperatur                     Format
# !!!!! SOLAX is only sending and updating data when powered by a PV panel,
# !!!!!!         otherwise get a lot of error meessage
#

########################
#
# SOLAX 1
#
########################

# SOLAX X1 mini G4 1.5K.
# ----------------------------------------------
#
#
#solax1localbegin    
  - platform: rest
    authentication: digest
    scan_interval: 15
    resource: http://@your IP adress /
    payload: "optType=ReadRealTimeData&pwd=@ yourSerialnumber of wifi Dongle to be included here"
    method: POST
    headers: 
      X-Forwarded-For: 5.8.8.8
    name: "solax1_rest_local"
[... 136 lines cut ...]
         icon: 'mdi:sine-wave'

       # compatible Solax X1 mini G4!
       - name: "Solax1 Inverter Nominal Power"
         #state: "{{ state_attr('sensor.solax1_local', 'Information')[0] | float | round(0) }}"
         state: "{{ state_attr('sensor.solax1_local', 'Information')[0] | float }}"
         unit_of_measurement: "kW"

       # compatible Solax X1 mini G4!
       - name: "Solax1 Module SN"
         state: "{{ state_attr('sensor.solax1_local', 'sn') }}"

        # compatible Solax X1 mini G4!
       - name: "Solax1 module fw version"
         state: "{{ state_attr('sensor.solax1_local', 'ver') }}"

       # compatible Solax X1 mini G4!
       - name: "Solax1 Inverter SN"
         state: "{{ state_attr('sensor.solax1_local', 'Information')[2] }}"

        # compatible Solax X1 mini G4!    
       - name: "Solax1 Inverter Information"
         state: "{{ state_attr('sensor.solax1_local', 'Information') }}"

#solax1localend

wills106, 2024-01-12:
I have added in your Inverter with [2024.01.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.6)

wills106, 2024-01-23:
Closing down as should be functional now, but no feedback.

closed 2024-01-23 by wills106