wills106/homeassistant-solax-modbus #639: [Bug]: Master Slave combined values not showing

meb31, 2023-11-24:
### Describe the bug

I am fairly new to HA. I have 2x X1 G4 Hybrids in master slave configuration but when I connect to the master via modbus it only shows the values in the single master inverter and does not account for the slave inverter's values hence my reported values are around 50% less than they should be. I have ticked the 'parallel mode' box on the configuration page, but this does not seem to do anything?
The only way I can get it to read the slave values is by adding another device with a different name but then I have twice the number of entities to deal with and for a lot of integrations this does not work.

### Integration Version

2023.11.4

### Homeassistant core version

2023.11.3

### Inverter brand

Solax G4 X1

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H4502AI40

### Inverter firmware versions

Latest

### Connection Method

Modbus TCP Waveshare

### Dongle firmware

Waveshare Modbus TCP

wills106, 2023-11-27:
When you tick "Parallel Mode" you should get Sensors starting "PM"
Also do you have the Inverters on a different address? https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-multiple-inverters-in-master-slave-mode-and-i-cant-connect-to-them-via-modbus

meb31, 2023-11-27:
Thanks Will for your sterling contribution in developing this integration.

Yes, both inverters have different modbus addresses. I spent a good couple of weeks figuring out why the inverter was coming back as 'unknown inverter' until I saw a post recently addressing this (pun intended).

I do have sensors with PM in the title but some like - sensor.solax_pm_activepower_l1 is only measuring the power of one of the inverters (the master I think). What is the difference between sensor.solax_pm_activepower_l1 and sensor.solax_pm_i2_activepower_l1 ?

wills106, 2024-02-18:
According to the docs "PM ActivePower L1" should be for all Inverters combined and any PM sensor with I2 in the name should be the values of the first slave Inverter like "PM I2 ActivePower L1"
I used I2 as in Inverter 2.

Unfortunately I only have a single Inverter and it's an older one that doesn't support Parallel Mode. So I can't perform any testing myself.

meb31, 2024-02-25:
> According to the docs "PM ActivePower L1" should be for all Inverters combined and any PM sensor with I2 in the name should be the values of the first slave Inverter like "PM I2 ActivePower L1" I [...]

Thanks for the reply. What documentation are you referring to? I could not find anything about this..

closed 2025-10-04 by github-actions[bot]