wills106/homeassistant-solax-modbus #641: Request about 5 more important Registers for actual SolaX Modbus Plugin

Stepsilein, 2023-11-26:
### Describe the bug

Hi wills106,
I donated yesterday and send you my wish/beg about adding 5 missing Registers at the Solax Modbus Addon.
Here again my message:

Hi wills106,
great job you did :-)
I just donated and asked you for kindely include 5 more very important registers at your famouse "SolaX Modbus" Addon.
Thanks a lot in advance.
(I´m German, so please excuse my bad English)
Sencerely, Stephan (Steps)
Here again the needed registers:
**0x00BF BMS_UserSOH 
0x00BA Battery_Tem_High
0x00BB Battery_Tem_Low
0x00BC Cell_Voltage_High
0x00BD Cell_Voltage_Low**

### Integration Version

? where to find?

### Homeassistant core version

2023.11.3

### Inverter brand

Qcells/Solax

### Plugin used

plugin_solax.py

### First 6 digits of Serial

H34C10

### Inverter firmware versions

33 09 31

### Connection Method

Wlan Dongle (working nice with your Addon) 

### Dongle firmware

Newest (Battery all = 3.15)
Inverter = 33 09 31

### Detailed Error Log

Nothing, but found now better way to ask you for including those 5 Registers, sorry.

### Additional context

Please send me any infos/answers to:
steps-post@gmx.de
Thans a lot in advance

cewbdex, 2023-11-26:
I am not the person you send money to, but I can implement this for you. However I would need an assistance from you. What do those entities do? I suppose historical max and min? What does SOH stand for? I need to know this to name them accordingly.

Stepsilein, 2023-11-26:
Hi,
I donated to wills106 by his donation link.
You are not wills106?

Anyway SOH = state of health in %.

Other 4 registers are low and high themperature in °C
 and low and high voltage in V (example: 3.42 V)

All 5 registers belong to the battery.

Bye for now.
Stephan

cewbdex, 2023-11-26:
I am not Wills, you can see my name above my comment. Will is an author of this integration, I am just another contributor. How experienced user are you? I could share a patch with you to test, these are sensor values, so there should be no risk of breaking anything.

cewbdex, 2023-11-26:
You can test changes from #642

Stepsilein, 2023-11-26:
Hi, me again.
You fot proving you, I donated wills106, I enclose as an attachment a few screens.

You are very kind to implement thos 5 registers/entities.
Thanks in advance.
Stephan
Attachements: 5 Images

Stepsilein, 2023-11-26:
Thanks a lot.
How di I update to your new version?
I use Home Assistant System at Raspberry Pi 400 with this Addon.

cewbdex, 2023-11-26:
I see you use email reply, it did not attach the images you sent. If these are proof of donation, I don't need them, I don't get any profit of this. You can try to test it with manually updating your files (either via SSH or you can use File Editor add-on, navigate to custom_components/... and edit insert the lines to the same position as in the Pull Request). If you don't feel brave for this, you can wait until Will merges them into the main branch and makes an update for the integration.

Stepsilein, 2023-11-26:
I just startet 3 weeks ago first time using HA and this addon.
So I know a few things.
About your link you sent already:
How to update now my system?
Where may I find a download?
Is it enough, to just exchange the solax.py file, to get thosd 5 register available, which you put in now?

Stepsilein, 2023-11-26:
Thanks again.
I tried 2 days ago implementing it at solax.py file, but didn't get it working.
So what should I put in at which file and so on?

cewbdex, 2023-11-26:
Visit the Changed files tab <https://github.com/wills106/homeassistant-solax-modbus/pull/642/files>

there you can see what and where is changed. Copy the changes or download the changed file and apply it to your installation. I have made the entities disabled by default, so either delete the line that makes each disabled or you can enable them manually later.

Stepsilein, 2023-11-26:
Thanks a lot.
I will give it a try later because I have to download via PC, Smartphone just saves .txt file.
I send you later info about my success 😉

Stepsilein, 2023-11-26:
Hi Kryštof,
I copied the canged 5 Entities into the plugin_solax.py in the right folder at HA.... but no way so see those 5 new entities to show up even after new start of HA, whole system, Browser and so on ;-(
What I´m still doing wrong ?
Here a screen of my solax.py now:
[image]

cewbdex, 2023-11-26:
Did you look for disabled entities? If not, try, or remove the `entity_registry_enabled_default` lines.

Stepsilein, 2023-11-26:
I did and no way get it working even deleted lines, or set it from false to true... sorry.
There is still something not working. 😢

The registers are working in Node-Red for a test.
For sure I disabled Node-Red after testing (was just to see, if registers are right).

cewbdex, 2023-11-27:
I don't know what is wrong there, maybe my patch is wrong, do you get any error messages? The entities do not show up at all, or are just zero? Correct way is to make changes, save, restart your Homeassistant core (or whole machine, but that is slower). Then you go to solax device page and entities should be there.

cewbdex, 2023-11-27:
Could you show me the values of the entities? I would like to see if they have valid values.

Stepsilein, 2023-11-27:
Give me some time.
I will include it in a dasboard and send yor screens about its values soon.

Stepsilein, 2023-11-28:
THANKS A LOT again.

cewbdex, 2023-11-28:
You are welcome, however, those values seem a bit odd to me, are you sure they correspond with reality? 4 ˚C seems very cold to me for batteries, or are they outside? Does SOH change over time, or they just hardcoded the 100 %? Also not clear to me what cell voltage does it measure? I have found SolaX Gen 2 has SOH already implemented, so I have refactored G4 to use the same name and icon, so don't be surprised after an update that the entity vanishes, it will have a different name. I guess I should add the rounding too, 3 digits work for you then?

Stepsilein, 2023-11-28:
Hi,
all values are realistic.
Batterys are stored unde the non insulated roof inside, but its very close to outside temperature.

SOH must be 100% as batteries are from 7/2023.. and hopefully will stay as long by 100%... may be after some years, it will drop a little.

Cell voltage is most important.
Its the 2 values, upper/lower in V from.my 2 battery packs.
It increases and degreases while charged/discharged.
If its like a paralell line/kurve, its okay.
If its driving away from each other, its an indication of imballence of some cells, which results in less usable charge capacity soon.
So its some 5 nice values now, to monitor battery health.

Temp. upper/lower simmilar.
Should't go below 0... but there ist battery heating included to guard against it.
Just monitoring is always safe stuff 😉.

The rounding is important too, as those values are just showing differences often at 2. digit after the , (or .).
You may include it, but some persons at the forum (not github) and I, did it already for us.

About future Updates of the Addon, I will have a look at the changeloc first.
Most new stuff isn't for SolaX often and so just a small part will bring me benefits to my setup.
But thanks for telling about.
What about wills106... is he still alive, don't read/see him... and never got some answers from him ... even not for my donation in the past 🫤.

About SolaX registers/sensors/entinities:
Are you working with that mid 2022 Somax Modbus pdf, listing all those at 52 pages?
If you know/have a newer one, please share it with me.
Thanks in advance 🙋🏼‍♂️

cewbdex, 2023-11-28:
My work is then complete. I would say most features of SolaX G4 are added when someone asks for them, as it's not very viable to implement everything, as there are no function descriptions in the document. I use probably the same V3.21 document as you. I don't know Will any better than you do, I would guess he is busy with his own life. IMHO donations are a gift, not a contract, he could help you when he gets a free while. Do you have any other questions?

Stepsilein, 2023-11-28:
> I am your neighbour from Czech republic, if that feeds your curiosity. But not a great weather for solar production here now.

Perfekt.
Here we have 15 cm snow today 🥴.

closed 2023-12-01 by wills106

Stepsilein, 2023-12-13:
tycto Entity ????
Anyway, did you use the newest plugind_solax.py ?
SOH should show about 100.
Batterie capacity should be about 6144 W if you use a 6 kWh System. (Simmilar to 9 ans 12 kWh)

matomatusov, 2023-12-13:
[Stepsilein](https://github.com/Stepsilein)
yes I have solax inverter modbus integration 2023.12.4.
I don't know why the given entities are null.

[image]

Stepsilein, 2023-12-13:
[image]

matomatusov, 2023-12-13:
Hi,
you have no idea why i have
sensor.solax_battery_state_of_health (SOH) = 0???

thanks

PR #642 (2023-12-01): SolaX G4 - add battery SOH, low and high values of battery temperature + cell voltage
fixes #641