wills106/homeassistant-solax-modbus #644: [Bug]: Soafar Solar HYD 10KTL-3PH reading failed Batery_xxxx_4 parametres

paveltresnak, 2023-11-27:
### Describe the bug

My inverter does not response on parameters with Battery_xxx_4. 

Log info:
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:465
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 26. listopadu 2023 v 23:23:37 (2101 occurrences)
Last logged: 07:33:20

sofar: read failed at 0x61b: Battery_power_4
sofar: read failed at 0x61c: battery_temperature_4
sofar: read failed at 0x61d: battery_capacity_charge_4
sofar: read failed at 0x61e: battery_state_of_health_4
sofar: read failed at 0x61f: battery_charge_cycle_4

### Integration Version

2023.11.4

### Homeassistant core version

2023.11.3

### Inverter brand

Sofar Solar

### Plugin used

plugin_sofar.py

### First 6 digits of Serial (Growatt Firmware Version)

SP1ES1

### Connection Method

Waveshare RS485 to ETH

cschlipf, 2024-02-19:
Are you sure that you have 4 Batteries connected? Which inverter model do you have? E.g. the HYD models just have two battery inputs.

paveltresnak, 2024-03-06:
No, I have connected only one battery. Sofar Solar HYD 10KTL-3PH have two batteries inputs.

cschlipf, 2024-03-06:
Which Battery Type have you installed and how many Battery modules?

paveltresnak, 2024-03-06:
I have one module Amass GTX 3000 with 5 battries.

cschlipf, 2024-03-06:
Can you successfully read "Battery Voltage 4" and "Battery Current 4"?

paveltresnak, 2024-03-06:
Batteries are in one Battery pack with one BMS. 
[home-assistant_2024-03-06T11-35-46.216Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14509230/home-assistant_2024-03-06T11-35-46.216Z.log)

cschlipf, 2024-03-06:
From your log I see way more read errors across the bank, which also affect well tested registers. As you have connected the adapter to the RS485 bus, have you properly terminated the bus on the inverter side with a 120 Ohm resistor?

cschlipf, 2024-03-06:
I am pretty sure this is your problem then. Please terminate the RS485 bus properly.

@wills106 - I suggest to close this issue (I can't close it myself). If a proper termination does not help, the issue can be re-opened again.

wills106, 2024-05-18:
Closing down as no further response in past couple of months.

Feel free to reopen if needed.

closed 2024-05-18 by wills106