wills106/homeassistant-solax-modbus #646: [Bug]: Unable to find entities

jacobsdad, 2023-11-27:
### Describe the bug

I have been unable to find additional entities. I followed the instructions for "Go to the SolaX Integration page. > >Find "+xy entities not shown", etc., but I am unable to find these options anywhere.

### Integration Version

2023.11.3

### Homeassistant core version

2023.11.3

### Inverter brand

Solax G4

### Plugin used

plugin_growatt.py

### Serial prefix

SRYBUL

### Connection Method

Pocket wifi

### Additional context

No additional info

wills106, 2023-11-27:
If you go to:
Settings
Devices & Services
1 Device, under SolaX Inverter Modbus
[image]
You will see a couple of +number. In my case under Diagnostic it's +21 entities not shown
[image]
I also have about +50 Under the Sensors grouping.
The amount you have is determined by your Inverter type.

jacobsdad, 2023-11-27:
Thanks for your reply. I've looked under all the settings and have nothing like that available:

[image]
[image]

jacobsdad, 2023-11-27:
My inverter is X1-Hybrid-G4.

wills106, 2023-11-27:
You don't have a working Modbus connection to your Inverter.

https://github.com/wills106/homeassistant-solax-modbus#pocket-wifi-30-with-firmware-v300403-and-above-is-only-officially-supported-solax-only-mentions-gen4-hybrid-other-inverters-may-work

If you have an older PocketWiFi 2 or below they are not reliable.

If you have a PocketWiFi 3.0 you may require a Firmware update.
https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-firmware-update/#pocketwifi-30
SolaX can do this remotely for you, if you email them.

jacobsdad, 2023-11-27:
Thanks. I will contact them and let you know if there was any success. Appreciate your time.

jacobsdad, 2023-11-28:
Hello again,

I have heard back from Solax who have updated my pocketwifi to the latest version (please see below). However, I am still only seeing 4 entities. Is there something else I am missing?

[image]

lieglw, 2023-11-28:
Hi, have a similar issue on Core 2023.11.3, so removed Solax device from integration and than removed the integration from Hacs restored a back up 2023.11.2
After the Back up is installed, got the newest version from the integration. Now it works again.
For my issue I needed to the Core to be 2023.11.2

jacobsdad, 2023-11-29:
Have closed. Thanks for your help and support.

closed 2023-11-29 by jacobsdad