wills106/homeassistant-solax-modbus #647: [Bug]: Updating state for some entities

johnnyhero83, 2023-11-27:
### Describe the bug

I got this log:

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:885
First occurred: 22. listopadu 2023 v 07:36:51 (16 occurrences)
Last logged: 08:10:07

Updating state for number.solax_feedin_discharge_min_soc (<class 'custom_components.solax_modbus.number.SolaXModbusNumber'>) took 0.582 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Updating state for sensor.solax_today_s_import_energy (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.735 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Updating state for number.solax_battery_charge_max_current (<class 'custom_components.solax_modbus.number.SolaXModbusNumber'>) took 0.741 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Updating state for sensor.solax_measured_power_l2 (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.674 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
Updating state for sensor.solax_inverter_voltage_l2 (<class 'custom_components.solax_modbus.sensor.SolaXModbusSensor'>) took 0.680 seconds. Please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

It started to happening after latest Solax modbus intergation has been uploaded to HA.

Thank you for checking

### Integration Version

2023.11.4 (latest)

### Homeassistant core version

2023.10.5

### Inverter brand

Solax X3 Hybrid G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

DSP 1.33 ARM 1.31

### Connection Method

Solax dongle 3

### Dongle firmware

V3.001.02

### Detailed Error Log

I checked detailed log and there is the same log information I am writing above. No further info

wills106, 2024-02-27:
There has been a number of changes in 2024.02.6 and [2024.02.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b2) to the way the Integration talks to the Inverter. Now using async and waiting for HA to fully load before establishing comms with the Inverter.

Hopefully the delays you mention have gone away now.

wills106, 2024-05-18:
Closing down as no further response in past couple of months.

Feel free to reopen if needed.

closed 2024-05-18 by wills106