wills106/homeassistant-solax-modbus #658: [Bug]: button press error

josefkrtil, 2023-12-06:
### Describe the bug

Update TOU charge times makes error. TOU start and end times stay - do not revert back

### Integration Version

2023.12.4.

### Homeassistant core version

2023.11.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES1

### Connection Method

RS485 - USB

### Detailed Error Log

2023-12-06 09:45:56.921 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written tou_id: 0.0
2023-12-06 09:46:04.642 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written tou_id: 3.0
2023-12-06 09:46:04.845 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [547439529920] None
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 230, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2035, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2072, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 235, in handle_service
    return await service.entity_service_call(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 876, in entity_service_call
    response_data = await _handle_entity_call(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 948, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 124, in _async_press_action
    await self.async_press()
  File "/config/custom_components/solax_modbus/button.py", line 86, in async_press
    if res: self._hub.write_registers_multi(unit=self._modbus_addr, address=self._register, payload=res)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 412, in write_registers_multi
    if hasattr(descr, 'reverse_option_dict'): value = descr.reverse_option_dict[value] # string to int
                                                      ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^
KeyError: None

wills106, 2023-12-08:
> TOU start and end times stay - do not revert back

The selects wont, they are different to the Numbers. I probably didn't explain that very well.
Only the number revert back after 120s if you don't write the values or the write fails, as they are coupled together correctly. Selects need further work.

I'll have another look at the value_function to see what's preventing it from writing.

cschlipf, 2024-02-19:
Wrong amount of registers are written, so it can't work as of today.

@josefkrtil - I would be interested in learning more about your use case what you want to achieve here. In my opinion the time of use mode has little value as it just provides very limited automation capabilities compared to what Home Assistant is offering. Why fiddle around with the complex TIme of Use modes, when you can easily create automations with the passive mode?

josefkrtil, 2024-02-20:
@cschlipf My idea was to use semi automatic operation. In case, when HA is not working, the inverter will work in its own rules. Now I am using full automatic operation with HA and passive mode.

cschlipf, 2024-02-20:
@josefkrtil - thank you very much. My current plan is to remove the timing and time of use modes as they are not working anyway. Once we have a working solution, we can add them again later.

Judging from what you wrote supporting TOU through Home Assistant would not be an important priority to you right now, is that correct?

josefkrtil, 2024-02-21:
@cschlipf yes, TOU is not priority for me.

cschlipf, 2024-02-21:
Thank you very much. Then I will use this issue to remove the TOU functionality first and re-introduce it later once it's working again.

PR #830 (2024-03-30): Removing Timing and TOU modes as they were non-functional.
Timing and Time-of-Use modes for Sofar did not work. This PR removes these non-functional entities and updates the documentation.

Closes #658
closed 2024-03-30 by wills106