wills106/homeassistant-solax-modbus #659: [Bug]: after update it doesn't work anymore Home assistant 2023.12

ipensieridellarchitetto, 2023-12-08:
### Describe the bug

[Bug]: after update it doesn't work anymore Home assistant 2023.12

### Integration Version

2023.10

### Homeassistant core version

2023.12

### Inverter brand

peimar

### Plugin used

plugin_growatt.py

### Serial prefix

000000

### Connection Method

wifi

### Additional context

after update it doesn't work anymore Home assistant 2023.12

wills106, 2023-12-08:
Could you please provide an error log. It's difficult to provide any trouble shooting with "it doesn't work anymore"

ipensieridellarchitetto, 2023-12-08:
ciao grazie sono riuscito a risolvere . ho reinstallato

-- 
*Flavio P.*
*Architetto*

*Visita i miei canali youtube:*
I pensieri dell'architetto
<https://www.youtube.com/channel/UCG7Fv9JKNa9NNHB_XGMYTDQ?sub_confirmation=1>

l'Impasto Magico
<https://www.youtube.com/channel/UCK_rdYQKUIuu5stzLwWbeVw?sub_confirmation=1>
e per i più piccoli il divertente Perrolino Channel
<https://www.youtube.com/channel/UCqONOt6jtc4P9-2NNzMENTw?sub_confirmation=1>

Seguimi su: INSTAGRAM
<https://www.instagram.com/i_pensieri_dell_architetto/>
                    FACEBOOK
<https://www.facebook.com/groups/405119710546941>

wills106, 2023-12-08:
Ok, I'll close the issue then.

closed 2023-12-08 by wills106