wills106/homeassistant-solax-modbus #661: Solis - Many entities have not state_class

i-jk, 2023-12-10:
### Describe the bug

Since 2023.12 the long-term statistics are visible via the History Graph. However, many useful Solis inverter stats like SOC and inverter current etc have no `state_class` set in the plugin file. This means that, though they have short term stats, no long term storage is kept for them.

See [my HA Community forum post on the subject](https://community.home-assistant.io/t/why-do-some-entities-have-no-statistics-entry/653096/4).

So, would you entertain a PR to add `state_class = SensorStateClass.MEASUREMENT` to all entities missing this config? Or shall we limit it to more important ones?

### Integration Version

2023.12.4

### Homeassistant core version

2023.12.1

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

not relevant

### Connection Method

not relevant

wills106, 2023-12-10:
Feel free to submit any PR's to improve the Integration.

I would recommend you start off adding `state_class = xyz` to entities you feel are important.

i-jk, 2023-12-11:
@wills106 ok will do... should I focus on input or holding registers? What modbus reg range am I aiming at?

i-jk, 2023-12-20:
@wills106 I went ahead and added the entities I wanted, basically anything that relates to performance. See https://github.com/wills106/homeassistant-solax-modbus/pull/676, thanks.

wills106, 2024-01-07:
Can this be closed down now?

wills106, 2024-01-10:
I'll close this down as the PR is released now.

closed 2024-01-10 by wills106