wills106/homeassistant-solax-modbus #665: [Bug]: Entity SolaX Battery State of Health and  BMS Battery Capacity

matomatusov, 2023-12-12:
### Describe the bug

Hi,
what could be the problem that it shows me zero values for 2 entities regarding the battery.
They are the following entities:
- SolaX Battery State of Health 
- BMS Battery Capacity

Thanks

[image]

### Integration Version

2023.12.4

### Homeassistant core version

core-2023.11.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.35 DSP1.34

### Connection Method

Waveshare adapter

### Dongle firmware

3.010

wills106, 2023-12-13:
I did mention in the Release notes that not all BMS's return values. [2023.12.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2023.12.1)

This is why I normally don't include any BMS registers as they have caused confusion in the past.

matomatusov, 2023-12-13:
HI,
I just that they are displayed to some and not to some. I am trying to understand why this is so.

Thanks

closed 2024-01-10 by wills106