wills106/homeassistant-solax-modbus #667: [Bug]: value of sensor.inverter_power in the same as inverter_power_l1 but not l1+l2+l3

gregorwolf1973, 2023-12-14:
### Describe the bug

The value of the sensor inverter_power is the same as inverter_power1 but it should be inverter_power1 + inverter_power2 + inverter_power3.
[image]
I think this also has impact on other values but i'm not sure.
- grid_import
- house_load

### Integration Version

2023.12.4

### Homeassistant core version

2023.12.2

### Inverter brand

SolaX X3 Hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H3PE05

### Inverter firmware versions

210117LC

### Connection Method

RJ45 TCP/IP Network

### Dongle firmware

none

martin-klima, 2024-01-29:
I just checked the sensors in my SolaX X3 Hybrid G3 installation and everything looks good.

[image]
[image]

gregorwolf1973, 2024-01-31:
Thanks!
I checked several times and reinstalled, rebooted. I have a X3 Hybrid G1/2 and here it is  different.
[image]

closed 2025-10-04 by github-actions[bot]