wills106/homeassistant-solax-modbus #669: Growatt SPH 4000 Register 1071 <-> 608

001Tycoon, 2023-12-16:
### Describe the bug

Hello, I have a Growatt SPH 4000 with Pylontech US2000C.
Single-Phase
Hybrid
RA1.0
Firmware RAAA212006 ZCBA-0006
Mode A0B1D0T0PFU2M4S7

For my inverters, “Battery Minimum Capacity” is not in register 1071 but in register 608. If I change the register in plugin_growatt.py it works. Unfortunately, I cannot say how to differentiate between the different inverter types. 

I would like an additional switch.
Holding Reg
Register NO. 00
Variable name OnOff
Description Remote on/off On(1);Off(0) inverter On(3) ;Off(2) BDC
Value 0,1,2,3
Note The inverter can be switched on and off, and the BDC can be switched on and off for the batt ready function.

Thank you for your work
Ingo

### Integration Version

2023.12.5

### Homeassistant core version

2023.12.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

RAAA21

### Connection Method

RS485 USR-W610

wills106, 2024-01-11:
What docs are you using the get the value of 608 from?

001Tycoon, 2024-01-11:
Hello,
it is described in a German forum.
https://www.photovoltaikforum.com/thread/192228-growatt-sph-modbus-rtu-rj45-pinout-und-register-beschreibung/?pageNo=2

choenig, 2024-01-11:
My PR https://github.com/wills106/homeassistant-solax-modbus/pull/697 also adds this register.

And, to be precise: 608 is `Load First Battery Minimum SOC` while 1071 is `Grid First Battery Minimum SOC` :)

wills106, 2024-01-12:
I have added the PR into [2024.01.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.6)

001Tycoon, 2024-01-12:
Thanks Register 608 works as it should

wills106, 2024-01-12:
I'll close this issue down then. Thanks for the PR @choenig

closed 2024-01-12 by wills106