wills106/homeassistant-solax-modbus #670: [Bug]: Stuck in waiting-mode on new X3-Hybrid-G4

mickemartinsson, 2023-12-16:
### Describe the bug

Got my old X3-Hybrid-G1/G2 (H3UE10) exchanged for a new X3-Hybrid-G4 (H34A15). There are some new interesting modes like manual mode. The problem is that the inverter goes to Waiting-mode (not idle) and when it does - it won't take any actions and any attempt to unlock won't work. Cant change mode (got 19% SOC) and whatever I do it just fails. 

### Integration Version

2023.12.5

### Homeassistant core version

2023.12.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

-

### Connection Method

Solax Pocket Wifi 

### Dongle firmware

-

wills106, 2023-12-16:
Unfortunately that's how they behave https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#unable-to-control-inverter-when-pv-0-battery-soc-is-at-minimum

You need to program a Charge Window, before the Inverter goes to sleep.

wills106, 2023-12-16:
We still haven't found a reliable way to awaken the Inverter once it goes to sleep.

mickemartinsson, 2023-12-16:
That was the case on the old inverter. This new G4 takes from the batteries when pv=0 (snow). Mine has been taken power from the batteries to keep it in normal mode, but after a while (still with >15% SOC) and after I have tried to change modes from HA it suddenly goes to Waiting mode. I should have kept taking from batteries to keep it normal but somehow it is now waiting for a test dump. Can this have to do with EPS and/or can a switch from grid to EPS make it normal again or is there a possibility to check why the test dump hangs?

wills106, 2023-12-16:
If you are getting an issue where the Inverter Hangs you might be on an older Firmware?

There is a link in https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-firmware-update/ for some Firmware info. You might be able to confirm if your Firmware is up to date? I can't verify the source of those updates. So I would advise contacting SolaX direct.

I don't own a Gen4 so I can't give anymore advice sorry.

Hopefully another Gen4 owner can shed some light on your problem?

mickemartinsson, 2023-12-16:
Thanks, I will check with the firmware and also talk to our local Solax dealer and try to find some more info on the problem.

wills106, 2023-12-16:
I am glad you still find this integration useful though!
It's been a while since we last collaborated.

mickemartinsson, 2023-12-17:
Absolutely, it's fantastic! I've used it in a research project (hopefully getting published soon), where it plays a crucial role in managing and optimizing the use of our solar cells and battery storage in conjunction with geothermal energy and electricity prices from Nordpool. Since 2018, we've managed to reduce our bought annual energy consumption from approximately 28,000 kWh to about 2,300 kWh without making any lifestyle changes. Thanks to the integration, I've also bought/sold energy at an average of over 25% better prices than if I had opted for the electricity companies' agreements based on monthly averages. The integration will receive a mention in the research report! So a big thank you for all the work you do with the integration!!

closed 2025-10-01 by github-actions[bot]