wills106/homeassistant-solax-modbus #671: [Bug]: Unlock Inverter & Unlock Inverter - Advanced

matomatusov, 2023-12-16:
### Describe the bug

Hi,
after the update Solax modbus (2023.12.5) stopped working two entities.

- Unlock Inverter
- Unlock Inverter - Advanced

Was it intentional?

[image]

Thanks

### Integration Version

2023.11.5

### Homeassistant core version

core-2023.11.3

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.35 DSP1.34

### Connection Method

Waveshare adapter

### Dongle firmware

3.010

cewbdex, 2023-12-16:
Yes, look in changelog. They were redundant.

matomatusov, 2023-12-16:
From this I did not understand that they were useless.

Thanks

closed 2023-12-16 by matomatusov

wills106, 2023-12-16:
I disabled the buttons as they are now a duplicate of the Select function. Which provides feedback which "Unlock" mode the Inverter is in.

matomatusov, 2023-12-16:
thanks, I understood.