wills106/homeassistant-solax-modbus #673: unrecognized Solis inverter type - serial number : unknown

MichaelNRW, 2023-12-18:
### Describe the bug

Hello,
I have the problem `unrecognized Solis inverter type - serial number: unknown`, although communication works with "normal" Modbus commands:

```
- name: EthernetGW2
  type: tcp
  port: 502
  host: 192.168.0.41
  delay: 5
  timeout: 5
  sensors:
```
    - name: modbus_wr_wirkleistung
      unit_of_measurement: W
      slave: 5                                                             
      address: 3004
      input_type: input                                                          
      data_type: uint32 
      precision: 2
    - name: modbus_wr_energie_total
      unit_of_measurement: kWh
      state_class: total_increasing
      device_class: energy
      slave: 5                                                             
      address: 3008
      input_type: input                                                           
      data_type: uint32 
      precision: 2
      scan_interval: 300
    - name: modbus_wr_energie_aktueller_monat
      unit_of_measurement: kWh
      state_class: total_increasing
      device_class: energy
      slave: 5                                                             
      address: 3010
      input_type: input                                                           
      data_type: uint32 
      precision: 2
      scan_interval: 300
    - name: modbus_wr_energie_letzter_monat
      unit_of_measurement: kWh
      state_class: total_increasing
      device_class: energy
      slave: 5                                                             
      address: 3012
      input_type: input                                                       
      #count: 2    
      data_type: uint32 
      precision: 2
      scan_interval: 300
(...)

The SOLIS S6-GR1P1K-M with the serial number 180101(...) is used.

### Integration Version

2023.12.5

### Homeassistant core version

2023.12.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

180101

### Connection Method

Waveshare RS485 to PoE ETH (B)

### Detailed Error Log

```
Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2319
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 11:18:20 (1 occurrences)
Last logged: 11:18:20

Solis: cannot find serial number, even not for other Inverter
```

```
Logger: custom_components.solax_modbus.plugin_solis
Source: custom_components/solax_modbus/plugin_solis.py:2339
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 11:18:20 (1 occurrences)
Last logged: 11:18:20

unrecognized Solis inverter type - serial number : unknown
```

wills106, 2023-12-19:
It looks like you are using the old Register set. Does plugin_solis_old return the Serial Number?

MichaelNRW, 2023-12-19:
```
2023-12-19 16:30:24.645 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis_old] Solis: cannot find serial number, even not for other Inverter
2023-12-19 16:30:24.645 ERROR (MainThread) [custom_components.solax_modbus.plugin_solis_old] unrecognized Solis inverter type - serial number : unknown
```

Asterix229, 2023-12-19:
Have same problem: ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : 067.01068.01 , with Growatt spf 5000es inverter , spf inverter is not listed, how to setup?

wills106, 2024-01-02:
@Asterix229 you don't have the same issue, can you post in https://github.com/wills106/homeassistant-solax-modbus/discussions/523 letting me know exactly what the Inverter is like PV only or Hybrid and amount of MPPT's etc

Asterix229, 2024-01-02:
Growatt SPF 5000 ES conected to local, usb-usb or rs485 modbus converter, work on pc program pvkeeper, on solaxpower recive some data if inverter put srne

Asterix229, 2024-01-02:
Growatt SPF 5000 ES conected to local, usb-usb or rs485 modbus converter, work on pc program pvkeeper, solaxpower recive some data if inverter put srne

Asterix229, 2024-01-02:
[Uploading OffGrid-Modbus-RS485RS232-RTU-Protocol-V0.14-20210420.pdf…]()

kupfi1212, 2024-01-11:
Hi, I have the same Problem with a QCell/Solax - Serialnumber: H34B10I2366295. The Connection is going over a WaveShare on the COM-Port to my Home Assistent. The Solar-System was build in September 2023.
_Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:66
Integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 21:03:42 (2 occurrences)
Last logged: 21:03:47
    SolaX: reading serial number from address 0x0 failed; other address may succeed
    SolaX: reading serial number from address 0x300 failed; other address may succeed_

wills106, 2024-01-23:
Closing down as no further response from original poster.

closed 2024-01-23 by wills106

kupfi1212, 2024-01-23:
Sorry for newcomer: Means this "closing down ...", that we have not found a solution and so the discussion is finiseh here?