wills106/homeassistant-solax-modbus #675: [Bug]: Solis - high volume RTC logs

durkimat, 2023-12-19:
### Describe the bug

I am getting a logbook entry for RTC time change upon every poll, which over time is causing large volumes of logs to be written.

example:
solis-modbus RTC changed to 2023-12-19 15:43:34_

I'm unsure if this is really a bug, but it is a nuisance as it also makes it difficult to see real events in the log due to the volume of noise created.

### Integration Version

2023.12.4

### Homeassistant core version

2023.12.3

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603105

### Connection Method

Solis S2-WL-ST

### Detailed Error Log

Error log is simply high volume of logs

solis-modbus RTC changed to 2023-12-19 15:43:34_

### Additional context

This may be intentional to create a log every 5 seconds (I have set polling every 5 seconds) but it seems to be strange for to have RTC logs every poll interval.

wills106, 2024-01-02:
I'm assuming you have enabled the sensor for RTC?
I normally leave it disabled and then just run an automation once a month or whatever to press the "Sync RTC" button.
Then you know it's always synced to your HA install.

wills106, 2024-01-10:
Closing down as not an issue.

closed 2024-01-10 by wills106