wills106/homeassistant-solax-modbus #678: [Bug]: only 1 entity is showing

markklos, 2023-12-26:
### Describe the bug

Only 1 entity is showing (serial number)

Log:
2023-12-26 12:16:43.873 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar_old] unrecognized Solar inverter type - serial number :  SB1ES140F9
2023-12-26 12:16:58.905 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

Type inverter: Sofar solar 4000TLM (3-5KTLM series)

### Integration Version

2023.12.6

### Homeassistant core version

2023.12.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar_old.py

### Serial prefix

SB1ES1

### Connection Method

rs485-usb

### Detailed Error Log

2023-12-26 12:43:05.210 ERROR (MainThread) [custom_components.solax_modbus.plugin_sofar_old] unrecognized Solar inverter type - serial number :  SB1ES140F9
2023-12-26 12:43:20.276 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

motclem, 2023-12-30:
I have a  similar situation, but I have 2 entities showing this is on a solar system

wills106, 2024-01-02:
Does this Beta load correctly? [2024.01.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.2b1)

wills106, 2024-01-03:
I have released a normal release [2024.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.2) with the same change.

If the Inverter is reading ok can you close this issue down.

wills106, 2024-01-07:
@markklos is your Inverter working ok now? Can I close this issue down now?

98jan, 2024-01-07:
Hi,

having same issue as the author:

```
2024-01-07 15:57:34.937 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-07 15:57:37.991 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-01-07 15:57:37.991 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2024-01-07 15:57:37.991 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
```

I have an X3 Gen4 15.0-D inverter from solax, it is connected through the COM port with my raspberry pi where the rs485 dongle is connected.
used the newest available release from home assistant and latest release from your github.

Any help is appreciated.

Best regards,
Jan

98jan, 2024-01-07:
my bad got it working now, was a mistake on my side. The unrecognized error messages could also be resolved with the adjustment to the homeassistant logger.

markklos, 2024-01-07:
> @markklos is your Inverter working ok now? Can I close this issue down now?

It seems to work, thanks.

closed 2024-01-07 by markklos